marazmista/radeon-profile #126: Trying to change power cap results in it being set to 1W

kieranc, 2019-03-05:
Everything else in the app works amazingly well, but whenever I try to set the power cap, no matter what value I choose, it gets set to 1W as displayed in the GPU summary window.
I'm using Kubuntu 18.10 with kernel 4.18.0-15, I have ROCm installed and rocm-smi can successfully set the power cap. Video card is a Vega 56. Radeon-profile 20190220.
Is there any other info I can provide to help debug this issue?

siyia2, 2019-03-06:
i can confirm the issue on debian 10 with rx550

marazmista, 2019-03-09:
Hi, possible fix in: 11b6870028359208af1b9a965a90fc776863fd8f

siyia2, 2019-03-09:
latest commit in combination with latest daemon fixed it thank you very much!!!

siyia2, 2019-03-09:
however power cap is not retained after a reboot compared to other settings

siyia2, 2019-03-09:
i found that in order to retain power cap after reboot you have to set the same power cap both at the default and custom profile

marazmista, 2019-03-10:
Do you have 'Restore selected overclock profile on start' checked in configuration?

siyia2, 2019-03-10:
yes but power cap only restores from default profile regardless

marazmista, 2019-03-10:
Ok, thanks for info, possible fix in 6d855fb3e068b9d3acc4de62beb61ed9ced35be8

siyia2, 2019-03-10:
power cap totally fixed thx, it would bebest to close the issue now

closed 2019-03-11 by marazmista