marazmista/radeon-profile #128: Low gpu load not being reported

siyia2, 2019-03-10:
Hello, gpu load when playing light games or videos and youtube is reported as 0%, running furmark gpu load shows at 100% as it should.

Kernel: 4.19.20
GPU: RX 550
OS: Debian 10

siyia2, 2019-03-13:
i checked with "watch cat /sys/class/drm/card0/device/gpu_busy_percent" and the gpu load behavior matches the radeon-profile behavior maybe it is a kernel-mesa issue

closed 2019-03-14 by siyia2

marazmista, 2019-03-14:
Hi, there is info about gpu load also in `amdgppu_pm_info` file in debugfs, but apparently the source of this value is the same as `gpu_busy_percent`. radeon-profile doesn't do any calculations here, just reads value. Here is some info: https://www.phoronix.com/forums/forum/linux-graphics-x-org-drivers/open-source-amd-linux/1031816-amdgpu-drm-driver-to-finally-expose-gpu-load-via-sysfs