marazmista/radeon-profile #129: Power Cap not saved after resume

siyia2, 2019-03-13:
Hello, after suspend or hibernate radeon-profile resets to max power cap,wether the limit is applied to default profile or to custom is irrelevant,one way to remedy that is to restart radeon-profile after resume from suspend.

siyia2, 2019-03-14:
doing also "echo 30000000 > /sys/class/drm/card0/device/hwmon/hwmon0/power1_cap" (for 30W in my case) after suspend works regardless of restarting the program or not.

jofr-well, 2021-03-19:
I think I have the same problem, but with manual frequency control. I set the manual control to not go beyond a certain level, but when Xorg's screen blank (screensaver) starts, this setting is reset and the card is no longer throttled. It only requires clicking "apply" to get the throttle back.

I'm trying to avoid disabling screen blanking for obvious reasons.