marazmista/radeon-profile #131: radeon-profile has greyed out options and I don't know what else to try

RavenChan1023, 2019-04-07:
As the title states, on radeon-profile, the overclock option is greyed out even after running with sudo, along with most other options relating to it and I don't know what's going wrong. Some people have said it may not work with amdgpu or something, but either way, I have a screenshot. Performance is absolutely abysmall, I really need this overclock but I will not modify the linux kernel in any way as that seems to be the only other option I can find.
[image]

parkerlreed, 2019-04-08:
You have to enable the feature mask for overclocking. 

https://wiki.archlinux.org/index.php/AMDGPU#Overclocking

Add that ppfeaturemask parameter to your kernel command line (in whichever bootloader you use)

marazmista, 2019-04-26:
Looks like your card is too old for `amdgpu` to work:
https://en.wikipedia.org/wiki/AMD_700_chipset_series#760G

closed 2019-04-27 by marazmista