marazmista/radeon-profile #133: Application claims my Vega 56 is running at 3-4ghz, despite only overclocking VRAM, only underclocking GPU

breakone9r, 2019-04-27:
I am running OpenSUSE Tumbleweed, and have boosted VRAM to 950mhz and undervolted the GPU to 1100mV on my Vega56 (and changed fan speeds to allow a 0 speed at idle temps) to keep noise and power levels down.

Imagine my surprise when I'm playing a game and I hear the GPU fans spool up. I tab out of the game and see that radeon-profile shows my CPU at 65C. which. isn't unheard of, but usually this game barely budges the GPU. Then I notice the GPU clock is 4063 MHz which is well above the clocks set at state 7. and bouncing between around 4100 and 3100 MHz, while my VRAM is bouncing betwen states 2 and 3.

These are pretty much the same settings that I used effectively in wattman. Under windows, the game in question wouldn't even cause a blip on the GPU usage, but it's pretty high here. (The game in question is NOT playing under wine or other emulation, but is available as a fully native linux game as well as windows and mac: Stellaris)

[image]

breakone9r, 2019-04-27:
Hm. After updating my kernel from 5.0.8 to 5.0.9 it seems the application is now giving normal readings for my GPU. Perhaps it was just a bug in the kernel?

marazmista, 2019-04-28:
Could be, although after a glance at changelog, for me, nothing obvious comes out that could have fix this issue  (http://lkml.iu.edu/hypermail/linux/kernel/1904.2/04945.html). Maybe libdrm update?

breakone9r, 2019-04-30:
Maybe... I only ever saw this issue this one time. I will go ahead and close it. If I see it again, I'll bring it back up.

closed 2019-04-30 by breakone9r