marazmista/radeon-profile #135: All buttons unclickable in Ubuntu 18.04

JC-S, 2019-06-03:
I installed using:
```
sudo add-apt-repository ppa:trebelnik-stefina/radeon-profile
sudo apt-get install radeon-profile
```
And then when I run ``radeon-profile``, the GUI popped up without problem, and shows the correct information about my GPU, but all buttons are grey and unclickable:
[image]

Dantali0n, 2019-06-03:
This bug also affects me, able to provide information about system and config if desired.

marazmista, 2019-06-03:
Do you have daemon installed?
Also, this seem to be very old version, that could be the cause.

JC-S, 2019-06-04:
> Do you have daemon installed?
> Also, this seem to be very old version, that could be the cause.

Thanks for your reply.
Yes, I do have a daemon installed. It is installed as a dependency by apt.

Also, is there any way to install a newer version other than compile from scratch? Is there any deb packages?

marazmista, 2019-06-04:
No, I don't think so. The only place I've found is this that you've mentioned in first post.

Enigma0, 2019-06-04:
@JC-S You can use this script to compile & install latest:

https://github.com/Enigma0/build-scripts/blob/master/build_radeon_profile.sh

Not a .deb but next best thing.

Marco-MP, 2019-06-05:
Actually I fixed this issue on my system by starting radeon-profile with root privileges

sudo radeon-profile

Dantali0n, 2019-06-05:
> Actually I fixed this issue on my system by starting radeon-profile with root privileges
> 
> sudo radeon-profile

I still experience the issue even when running with root privileges.

Mikasu-X14, 2019-07-13:
I got it working on 18.04 while having the daemon. Really makes it easy as it goes by the daemon to control the card rather than root. Just be sure you got the daemon installed. If it doesn't work still, I don't know, I guess you're unlucky

Thisisaline, 2019-07-21:
I have a similar issue on 18.04.
Some of the buttons and tabs work, some don't. Also, only the Connections tab has information, everything else is blank.

Mikasu-X14, 2019-07-23:
> I have a similar issue on 18.04.
> Some of the buttons and tabs work, some don't. Also, only the Connections tab has information, everything else is blank.
> Edit: After further testing I realise that the same thing happens when I run radeon-profile without the daemon. It seems I might be doing something wrong trying to run the daemon, but I don't know what.

I just realized that I compiled radeon-profile and it's daemon, I didn't install it through a ppa. Maybe try that. For both, you just got to do 'sudo make install' at the end.

Thisisaline, 2019-07-26:
> I just realized that I compiled radeon-profile and it's daemon, I didn't install it through a ppa. Maybe try that. For both, you just got to do 'sudo make install' at the end.

I compiled them as well cos as far as I know there is only one ppa available for radeon-profile that is quite outdated. Also it is my understanding that sudo should not be necessary. I compiled both as per the instructions, with 'qmake && make'.
What could be the issue for me is that I have not moved the systemd service file to /etc/systemd/system/. I don't know this for sure though as I have not yet got around to doing it, and being new to linux, I first have to figure out how to do it.

Mikasu-X14, 2019-07-26:
@Thisisaline `sudo make install`
should do on both, it will install the app system wide and it should install the daemon correctly
then, with `sudo service --status-all` you can see if the daemon is started (if it has a plus, it's running). If it's not, just do `sudo service <service> start`

Thisisaline, 2019-07-26:
ok thanks, but isn't the tool supposed to be able to run without root access?

Mikasu-X14, 2019-07-26:
you're giving it root access so it can be installed on the system @Thisisaline . You will have to do it to install the daemon. For the software though, you can just run the compiled code like you were already doing. It wont run as root, even when installed on the system. Though, if your daemon is not installed, it cannot access the card. It's because it is going by the daemon to access what's not accessible without root.

Thisisaline, 2019-07-26:
ok, so even though I have already run `make` with the daemon I can just go into radeon-profile-daemon/radeon-profile-daemon folder and run `sudo make install`?
Do I need to copy the systemd service file to /etc/systemd/system/ like it mentions on the daemon git page? If yes, how do I do it?

Mikasu-X14, 2019-07-26:
@Thisisaline the `sudo make install` is for the daemon. Though, you can do it for radeon-profile if you want it to appear with it's icon like all of your other programs in the launch menu. I honestly don't know how to setup a service. I did it once for a TeamSpeak server but that's about it.

marazmista, 2019-07-27:
Hi. To install systemd service (after you've executed `make install` on deamon) you need to copy `radeon-profile-daemon.service` to `usr/lib/systemd/system/` and later execute `systemctl enable radeon-profile-deamon --now`. That should start the daemon.

Mikasu-X14, 2019-07-27:
I guess I was the only one that it installed automatically...
the service commands should do too.

Dantali0n, 2019-07-28:
I have installed radeon-profile by building it from source but no matter how I run it I can still not press any buttons

Mikasu-X14, 2019-07-28:
@Dantali0n did you install the daemon?

Dantali0n, 2019-07-28:
> @Dantali0n did you install the daemon?

No and I like not to, the documentation states it should work without 'Otherwise app need to be run with root privilages for changing power profiles (and clocks readings sometimes)'

Thisisaline, 2019-07-28:
@Dantali0n your phrasing suggests you're looking at it backwards.. you need the daemon to run radeon-profile without root privileges, not the other way round.

@Mikasu-X14 @marazmista Thanks. I only ran `make` and not `make install`, that's probably contributed to my situation. 
So I can run `make install` without `sudo`, or is `sudo` necessary? 
Also, sorry for a noob question, but I assume I can copy `radeon-profile-daemon.service` to `usr/lib/systemd/system/` by running `sudo cp radeon-profile-daemon.service usr/lib/systemd/system/radeon-profile-daemon.service` from within the extra folder, is this correct?
Also, can I enable and start the service file straight after doing the above, or do I have to somehow navigate to the `usr/lib/systemd/system/` folder first?
And if I may lastly ask for my own knowledge, since apparently running `systemctl enable radeon-profile-daemon.service` AND `systemctl start radeon-profile-daemon.service` vs running only `systemctl enable radeon-profile-deamon --now` both start the daemon, what is the difference between both methods? Is the `--now` command like a sort of alternative short cut for the `start` command? 

Thanks a lot!

Mikasu-X14, 2019-07-28:
@Thisaline 
Sudo is necessary because it installs on the system where only root can modify anything.
It's /usr . Always from root. Else, it wont work. 
For the command, I guess it's just to keep the legacy way of doing things. I think most people use service niwafays as opposed to systemctl.

Mikasu-X14, 2019-07-28:
@Dantali0n you will have to run with root then. If it's not working, maybe you are not on an up-to-date version or it didn't compile correctly, I can't tell

Dantali0n, 2019-07-29:
@Mikasu-X14 @Thisisaline I already stated I run radeon-profile as root on the 5th of June, further emphasized by 'no matter how I run it'. I think saying its not up-to-date or diden't compile correctly is something we should not immediately assume, especially since it was working for me before (and I did not reinstall).

I think it is likely there is a bug that is causing this problem and we should attempt to investigate.

Mikasu-X14, 2019-07-29:
@Dantali0n could also have to do with your kernel version but that's unlikely. Maybe communicate with the developper to see what can be done about your case.

Thisisaline, 2019-08-07:
I've got the tool mostly working now, except for the part I need the most, the overclock tab (since I need to underclock my gpu).
I've already enabled overdrive bit on grub when I installed wattmangtk, and as far as I understand, I did it correctly.
What should I do to get the overclock tab to work?