marazmista/radeon-profile #137: Manually choosing a memory clock to eliminate flicking

TeenCorn, 2019-06-07:
Allow the user to be able to choose a fixed memory clock. Setting **pp_dpm_mclk** to "2" completely removes the flickering that happens when I set the OverDrive bit on amdgpu.ppfeaturemask. Currently have to open a terminal and run a script to that for me :(

marazmista, 2019-06-26:
Yes, but first I have to modify UI for this feature because apparently, that file accepts multiple values (to enable/disable states) so list with checkboxes will fit better than current sliders (which allow only one state to be enabled)

TeenCorn, 2019-06-28:
Nice. Looking forward to the change. And thanks for the handy tool!

marazmista, 2019-08-25:
Implemented in 9a364be47b751165504362b49b38d011c6712636

closed 2019-09-03 by marazmista