marazmista/radeon-profile #142: SUDO  problem

yuceltoluyag, 2019-06-27:
without sudo, the entire frame is empty and greyed out.
```
sudo radeon-profile
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
```

work
[image]

parkerlreed, 2019-06-28:
Even doing nopasswd and the binary path, you still have to call sudo. That just makes it so it doesn't ask for password.

TeenCorn, 2019-06-28:
You have to install radeon-profile-daemon to run it as a normal user.

yuceltoluyag, 2019-06-28:
I set it up at first @TeenCorn 
```
systemctl status radeon-profile-daemon.service 
● radeon-profile-daemon.service - radeon-profile daemon
   Loaded: loaded (/usr/lib/systemd/system/radeon-profile-daemon.service; enabled; vendor preset: disabled)
   Active: active (running) since Fri 2019-06-28 21:54:04 +03; 33s ago
 Main PID: 508 (radeon-profile-)
    Tasks: 2 (limit: 4915)
   Memory: 10.1M
   CGroup: /system.slice/radeon-profile-daemon.service
           └─508 /usr/bin/radeon-profile-daemon

Jun 28 21:54:04 Baba systemd[1]: Started radeon-profile daemon.
Jun 28 21:54:07 Baba radeon-profile-daemon[508]: "radeon-profile-daemon (v. 20190603)"
Jun 28 21:54:07 Baba radeon-profile-daemon[508]: Awaiting connections...
```

yunta-mb, 2019-07-01:
Radeon-profile uses '/run/radeon-profile-daemon-server' socket to communicate with its daemon. You may want to check if that socket is visible & accessible by the user running radeon-profile.

closed 2019-07-01 by yuceltoluyag