marazmista/radeon-profile #143: Compilation error with Fedora 30

Tilt78, 2019-06-30:
Hello,
I have this error with Fedora 30:
[Code]
[ludo@arch-anywhere radeon-profile]$ qmake-qt5
Project MESSAGE: Building for release
[ludo@arch-anywhere radeon-profile]$ make
g++ -c -pipe -std=c++11 -O2 -Wall -W -D_REENTRANT -fPIC -DQT_NO_DEBUG_OUTPUT -DQT_NO_DEBUG -DQT_CHARTS_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -isystem /usr/include/qt5 -isystem /usr/include/qt5/QtCharts -isystem /usr/include/qt5/QtWidgets -isystem /usr/include/qt5/QtGui -isystem /usr/include/qt5/QtNetwork -isystem /usr/include/qt5/QtCore -I. -isystem /usr/include/libdrm -I. -I/usr/lib64/qt5/mkspecs/linux-g++ -o gpu.o gpu.cpp
gpu.cpp:12:10: erreur fatale: X11/extensions/Xrandr.h : No such file or directory
   12 | #include <X11/extensions/Xrandr.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminée.
make: *** [Makefile:712: gpu.o] Error 1
[/code]
Any ideas ?

yunta-mb, 2019-07-01:
try again after:
`sudo dnf install qt5-devel qt5-qtcharts qt5-qtcharts-devel libXrandr-devel`

Tilt78, 2019-07-01:
Thanks a lots, its installed, and perfectly working.
Great !

closed 2019-07-01 by Tilt78

jlaw9, 2020-05-14:
Had to run this on Ubuntu:
`sudo apt install libxrandr-dev`