marazmista/radeon-profile #146: Multiple tray icons

SinisterBrain, 2019-07-27:
Hi, first.  Thanks for the awesome program.  Very glad this exists.  

I just have a strange issue.  I'm using Manjaro KDE.  Oddly it seems to keep spawning multiple Readeon-Profile icons in the task tray. Only one of them actually displays information.  I have to right click to get dismiss them.
Of course, this could be a KDE issue.  I promise I also only have one radeon-profile and daemon installed.  ;)
I did a search and nothing showed but I'm notoriously bad at find issues on github.  So apologies if this is a repeat issue.
Also, it seems I'm one release behind.  The AUR one doesn't seem to have been updated for your current release?
`aur/radeon-profile-git 20190211.r0.g52f3ee3-1 (+60 4.56%) (Out-of-date 2019-07-08) (Installed: 20190603.r4.g981aee4-1)`

[image]

Thank you for your time!

Zamundaaa, 2019-08-06:
Does this happen on boot? Or are you opening radeon-profile from the task manager sometimes? Because that doesn't open the old instance but instead starts a new one. I'm not sure if that's wanted behaviour though.
I'm on Manjaro KDE too and there's definitely only one instance starting on boot.

SinisterBrain, 2019-08-07:
Hi, Thanks for replying.  
Yes the screenshot was taken a few minutes after booting.  
I rarely actually open the program.  I have it set to make the adjustments I want and I leave it be.  
I've tripled checked that I didn't somehow add it a bunch of times to the varies locations you can add an autostart including System settings in KDE.  
It's done it since I've installed it about a month ago.  But it seems to be getting worse.  At first it was 3 or 4, then 6 or 8, and now it's sometimes a dozen.  Only one actually shows the card info on mouseover.  The rest are blank so I right-click -> close them.

SinisterBrain, 2019-08-07:
Welp, I feel a little dumb.
I think I tracked this down to the session restore.  
There was something on the Manjaro forums that made me think of the session restore. 
So I set it to `Start with an empty session` and it hasn't began reproducing after a few reboots.
It definitely does it for me when restoring a previous session though.

Oxalin, 2020-03-12:
Maybe we should prevent starting more than one instance per session at any given time... unless someone identify a case where many instances would be needed.

Oxalin, 2020-03-24:
I'll prepare a patch and post it here if someone is willing to test it. It may take a few days. I'll be back soon.