marazmista/radeon-profile #151: "Save selected fan control mode" doesn't act as expected; fan profile not restored on startup.

Lomomn, 2019-08-15:
I assume this should restore the last fan setting that I selected when the app starts up again. However, this is not the case. Could you clarify what this setting actually does?

Is there a way to set my fan profile back to my custom curve at startup?

My fan will use the card bios/system default/auto in radeon-profile which allows it to reach >70C.

naglfar, 2019-08-16:
I think it is supposed to work like that, since at least in my case when I start the application my fan profile is clearly shown as selected on the bottom left and on the fan control tab. It's just that... it actually isn't. I have to switch to auto or fixed and then back to my profile to actually have it kick in. This also happens on wake up from sleep / suspend to ram.

Lomomn, 2019-09-17:
@marazmista Can you confirm what this setting should actually do? There is no title/hover text and I can't find any documentation for it.

marazmista, 2019-09-21:
If you have this option checked it should restore selected fan control on start as it was on exit. So if it was fixed speed, same should be restored on next start. If you have custom fan profiles, let's say A, B and C and B was selected before close of application, this is saved in config file and B should be restored on next start.

Lomomn, 2019-09-21:
I made my own fan profile, which is saved under name "default".
When the program opens up (automatically on startup) the fan profile is set back to "auto".

If I close the program fully and reopen, it still uses "auto" fan profile.

Is there anything I can do to help debug this?

Using version 20190903.

naglfar, 2019-09-23:
So, I didn't get around to commenting again, but 20190903 actually seems to have fixed this issue for me, meaning my selected fan profile actually is active on program start. It still doesn't work on wake up after sleep / suspend, but for now I just work around this using a script that kills and restarts the gui.

Lomomn, 2019-09-23:
Weird. When I close and re-open the program the fan profile is back to "auto." 
Are you using the daemon too?

naglfar, 2019-09-24:
Yes, daemon is running and the gui as normal user.

marazmista, 2019-09-24:
Apply profile after wake up / suspend is difficult one because, as far as I know, app doesn't know that system just woke up. Solution for this could be restart of gui using udev event or script, just like you said.
You can verify whether setting has been saved on exit by looking into config file: `~/.config/radeon-profile/radeon-profile-settings` and there, if `fanMode` equals `2` that means custom fan profile. `fanProfileName` is the name of active profile on exit.

Lomomn, 2019-09-24:
fanMode never changes in the config file. It is always `fanMode=0`.
When I change mode in GUI, right click on the appindicator and press quit, it defaults back to auto fan mode.

chown'ing the config file and manually changing to `fanMode=2` does retain the setting, however.

I guess this is fixed now, since the file should have had user perms when it was made. If other people could change their config using the app, that would mean that they had perms.

Oxalin, 2020-03-06:
@marazmista This issue can be closed (wrong permissions on setting file).

However, for the suspend/wake up problem where the fan profile will not be kept, would there be a way to check if the fan mode was not preserved?

marazmista, 2020-03-27:
About suspend/wake up. Solution to this could be, for example, checking from time to time whether value read from sensors about fan speed and temperature corresponds to values defined in fan profile. When they don't match, try to apply profile again.

closed 2020-03-27 by marazmista

Oxalin, 2020-03-30:
Yes, this could work.

supiiik, 2020-06-04:
I have same problem with fan profile after wake up. The fan is using a different profile (it's not default nor auto profile). But I can see, that the right profile is selected, the RPMs of the fan doesn't correspond to the select profile curve.
I'm using 20190903 version. Only when I change to fixed or auto and back to custom curve, the fan starts to follow the selected curve

FMeinicke, 2020-09-12:
Hi, so I also experience the issue that my custom fan profile is not used after waking up from suspend.  

I started digging around a bit and found that it's actually not an issue with radeon-profile. At least on my system (don't know if this is a phenomenon related to all Linuxes or if it just happens on Arch/Manjaro) when the PC is sent to sleep the value for pwm_enable is reset to 'auto' again. Hence, when the system wakes up, setting new PWM values doesn't actually change the fan speed.

The problem is that radeon-profile doesn't know that the system has just woken up from sleep (as someone already mentioned above) so it continues to send PWM values believing that these will actually set the correct fan speed. What I tried to fix this is to add a check before setting a new value to see if the correct mode is set for pwm_enable. If not, the mode is set to 'manual' if it was before going to sleep.

If someone is interested in the code, I'm happy to open a PR.

marazmista, 2020-09-13:
> If someone is interested in the code, I'm happy to open a PR.

@FMeinicke sure, please do. I think this is a great idea and it would be good to check if pwm_enable has correct value before setting fan speed.

Oxalin, 2023-01-24:
Fixed under my fork since the development seems to have halted. Thank you.