marazmista/radeon-profile #152: High Pitch Fan Sound

RickAndTired, 2019-08-21:
Radeon-Profile is causing my RX 570 fan to make a loud high pitch sound after quitting.

System:
Ubuntu 18.04
Kernel 5.0.0-25
Sapphire Radeon RX 570
Driver: Mesa 19.1.0+aco+git1566305973-9b18c4e3df0-1bionic1
Radeon-Profile: v. 20190603

Scenario 1:
Normal use without radeon-profile does not produce the high pitch sound
Starting radeon-profile does not produce the high pitch sound
Quitting radeon-profile causes the fan to produce the high pitch sound
(nothing was touched in radeon-profile)

Scenario 2:
Normal use without radeon-profile does not produce the high pitch sound
Starting radeon-profile does not produce the high pitch sound
Changing fan settings to 100% (or any %) does not produce the high pitch sound
Changing the fan back to Auto causes the fan to produce the high pitch sound
Changing the fan to anything other than Auto will remove the high pitch sound
Quitting radeon-profile causes the fan to produce the high pitch sound

marazmista, 2019-08-21:
In nominal conditions, when there is no high pitch sound, don't start radeon-profile and try to change fan mode manually from console:
`echo "1" > /sys/class/drm/card0/device/hwmon/hwmon0/pwm1_enable`
and later, go back to auto:
`echo "2" > /sys/class/drm/card0/device/hwmon/hwmon0/pwm1_enable`

Does this cause high pitch sound?

RickAndTired, 2019-08-21:
Yes it does
For hwmon2, I don't have an hwmon0 directory there

marazmista, 2019-08-25:
So the problem is likely somewhere deeper, maybe in kernel.

Podesta, 2019-09-02:
Yeah, same issue on RX580. I had reported it back on another thread, since it was worse when the the daemon would disconnect from the profile quite often. Once it starts making the pitch noise, if you open the profile again, you can see that it reports the fan all over the place, going up and down quickly. I usually fixed it by setting the fan to "auto" and selecting the "profile_peak" setting, which maxes the fan speed, and then you can revert to whatever settings you had.

Oxalin, 2020-03-06:
@RickAndTired is this still relevent? From your answer to comment https://github.com/marazmista/radeon-profile/issues/152#issuecomment-523573045, your problem doesn't come from Radeon-profile per se, but from your kernel when switched in auto mode. If this is still happening (through or without radeon-profile), you should open a bug on the kernel side. Also, you could read values the values from hwmon interfaces for GPU fan (see here) before and after triggering the high pitch sound,