marazmista/radeon-profile #153: Control of acceleration.

Ukol1, 2019-08-21:
Thank you very much for the AMD graphics program. I use Manjaro Linux x86_64 Gnome version 2019.06.03. Gnome does not fully control the panel as it does for KDE.
On the overclocking tab : GPU overclocking works, memory does not work. When you select Manual frequency control, the change sliders do not work, States table is not active.
P.S Sorry for the translation.
[image]
[image]
[image]

marazmista, 2019-08-21:
Hi, some info about manual frequency control: https://github.com/marazmista/radeon-profile/issues/137

About the states table being disabled, I see that you have mobile version of card, do you have `pp_od_clk_voltage` file in `/sys/class/drm/card0/device/`?

Also your core clock look odd but I'm not sure what cause that.

Ukol1, 2019-08-21:
I don't have that file. The frequency of the GPU depend on the inclusion of acceleration.
[image]
[image]

Ukol1, 2019-08-22:
I loaded a LiveCD of Ubuntu 18.10. There is a version of the program 2018.10.28, it's a simple comparison for overclocking the video card. There I, too, such a file not found.
[image]
[image]

marazmista, 2019-08-25:
If you don't have this file, States table won't work.
I've pushed implementation of Manual frequency control: 9a364be47b751165504362b49b38d011c6712636

Ukol1, 2019-08-26:
Hey. Good job. I checked all the GPU overclocking modes they work. Memory overclocking still doesn't work for me. When you change the slider in the file pp_mclk_od changes occur.
[image]
[image]

marazmista, 2019-08-26:
Does memory overclock work if you change it by hand from console?

Ukol1, 2019-08-27:
Yes if you change the value in pp_mclk_od, the changes are applied after you restart value becomes the default.

Ukol1, 2019-08-28:
Hello again. Is it possible to add a separate power management?! I have auto, high performance or manual, the power rises to 1.2 volts, artefacts appear or even the image is gone. When you choose low power 0.9 volts, the card works fine but no performance, frequency I cannot change. Any thoughts on downvoting your graphics card?

Ukol1, 2019-08-28:
Hey. Here's a picture where I showed what works or doesn't work correctly. Does not change the memory frequency, constantly on default settings 945MHz.
[image]

marazmista, 2019-08-29:
Memory overclock fixed in 93fb5cb12069bc0a0531cbc007462b5f7833b53b and values in table update 519703783aa1ad38dd04aa922a45b4106683b13f

Ukol1, 2019-08-29:
Hi you did a great job! Everything works as it should. Attach screenshots. Thanks again!
[image]
[image]

Ukol1, 2019-08-30:
Hi! Have a request, when you update the package to change the numbering of the Assembly. My main system Rosa Fresh R11 Plasma. Everything works fine. Thanks!
[image]
[image]

marazmista, 2019-08-31:
Thanks for testing.

Ukol1, 2019-08-31:
Thank you for the program! Keep up the good work!

closed 2019-09-03 by marazmista