marazmista/radeon-profile #157: Issues with Navi cards

shmerl, 2019-09-04:
I tried using radeon-profile with Navi (Sapphire Pulse RX 5700 XT), kernel 5.3-rc5. Mostly for the purpose of defining a custom fan curve. It works, but intermittently I see such messages in dmesg:

``` 1871.985312] amdgpu: [powerplay] Failed to send message 0xe, response 0xfffffffb param 0x80
[ 1871.985315] amdgpu: [powerplay] Failed to export SMU metrics table!
[ 1881.985445] amdgpu: [powerplay] Failed to send message 0xf, response 0xfffffffb, param 0xa90000
[ 1881.985460] amdgpu: [powerplay] Failed to send message 0x12, response 0xfffffffb, param 0x6
[ 1883.985641] amdgpu: [powerplay] Failed to send message 0xf, response 0xfffffffb, param 0xa90000
[ 1887.985544] amdgpu: [powerplay] Failed to send message 0xf, response 0xfffffffb, param 0xa90000
```

Once that occurs, GPU can also hang (randomly).

shmerl, 2019-09-04:
Also, traces like this happen:

```[  147.924191] WARNING: CPU: 3 PID: 4396 at drivers/gpu/drm/amd/amdgpu/../display/dc/dcn20/dcn20_hwseq.c:1942 dcn20_setup_gsl_group_as_lock+0x8f/0x210 [amdgpu]
[  147.924192] Modules linked in: rfcomm(E) nf_tables(E) nfnetlink(E) bnep(E) edac_mce_amd(E) kvm_amd(E) kvm(E) irqbypass(E) nls_ascii(E) nls_cp437(E) vfat(E) fat(E) crct10dif_pclmul(E) crc32_pclmul(E) ghash_clmulni_intel(E) btusb(E) btrtl(E) btbcm(E) btintel(E) bluetooth(E) uvcvideo(E) videobuf2_vmalloc(E) iwlmvm(E) drbg(E) videobuf2_memops(E) videobuf2_v4l2(E) videobuf2_common(E) snd_hda_codec_realtek(E) aesni_intel(E) ansi_cprng(E) mac80211(E) aes_x86_64(E) snd_hda_codec_generic(E) libarc4(E) crypto_simd(E) ledtrig_audio(E) videodev(E) ecdh_generic(E) snd_hda_codec_hdmi(E) ecc(E) iwlwifi(E) crc16(E) snd_usb_audio(E) snd_usbmidi_lib(E) snd_rawmidi(E) sg(E) snd_seq_device(E) efi_pstore(E) cryptd(E) mc(E) wmi_bmof(E) glue_helper(E) snd_hda_intel(E) cfg80211(E) pcspkr(E) rfkill(E) snd_hda_codec(E) sp5100_tco(E) snd_hda_core(E) efivars(E) snd_hwdep(E) snd_pcm(E) watchdog(E) snd_timer(E) ccp(E) snd(E) soundcore(E) rng_core(E) evdev(E) acpi_cpufreq(E) nct6775(E) hwmon_vid(E) parport_pc(E)
[  147.924210]  ppdev(E) lp(E) parport(E) efivarfs(E) ip_tables(E) x_tables(E) autofs4(E) xfs(E) btrfs(E) xor(E) zstd_decompress(E) zstd_compress(E) raid6_pq(E) libcrc32c(E) crc32c_generic(E) sd_mod(E) hid_generic(E) usbhid(E) hid(E) amdgpu(E) gpu_sched(E) ttm(E) drm_kms_helper(E) xhci_pci(E) ahci(E) libahci(E) mxm_wmi(E) xhci_hcd(E) drm(E) libata(E) crc32c_intel(E) i2c_piix4(E) mfd_core(E) scsi_mod(E) igb(E) usbcore(E) dca(E) usb_common(E) ptp(E) nvme(E) pps_core(E) i2c_algo_bit(E) nvme_core(E) wmi(E) button(E)
[  147.924221] CPU: 3 PID: 4396 Comm: Xorg Tainted: G        W   E     5.3.0-rc5-amd64 #1 Debian 5.3~rc5-1~exp2
[  147.924222] Hardware name: To Be Filled By O.E.M. To Be Filled By O.E.M./X570 Taichi, BIOS P2.00 08/21/2019
[  147.924294] RIP: 0010:dcn20_setup_gsl_group_as_lock+0x8f/0x210 [amdgpu]
[  147.924296] Code: b6 80 70 02 00 00 a8 01 0f 84 3b 01 00 00 a8 02 0f 84 ea 00 00 00 a8 04 0f 84 4f 01 00 00 48 c7 c7 00 16 9b c0 e8 af 8e 08 e5 <0f> 0b c6 83 60 02 00 00 00 48 c7 c7 00 16 9b c0 e8 9a 8e 08 e5 0f
[  147.924297] RSP: 0018:ffff9cae0971f9c0 EFLAGS: 00010246
[  147.924298] RAX: 0000000000000024 RBX: ffff8c04511101b8 RCX: 0000000000000006
[  147.924298] RDX: 0000000000000000 RSI: 0000000000000086 RDI: ffff8c047e6d7680
[  147.924299] RBP: 0000000000000001 R08: 0000000000000a86 R09: 0000000000000004
[  147.924300] R10: 0000000000000000 R11: 0000000000000001 R12: ffff8c046a5c0000
[  147.924301] R13: ffff8c0451111d18 R14: ffff8c04511101b8 R15: ffff8c04274c6200
[  147.924302] FS:  00007f12a79e3f00(0000) GS:ffff8c047e6c0000(0000) knlGS:0000000000000000
[  147.924303] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  147.924303] CR2: 00007f128043e000 CR3: 00000007d6c20000 CR4: 0000000000340ee0
[  147.924304] Call Trace:
[  147.924375]  dcn20_pipe_control_lock.part.19+0xe4/0x1b0 [amdgpu]
[  147.924443]  dc_stream_set_cursor_attributes+0x11f/0x170 [amdgpu]
[  147.924518]  handle_cursor_update.isra.50+0x1b2/0x310 [amdgpu]
[  147.924526]  drm_atomic_helper_async_commit+0x63/0xd0 [drm_kms_helper]
[  147.924535]  drm_atomic_helper_commit+0xc4/0x120 [drm_kms_helper]
[  147.924541]  drm_atomic_helper_update_plane+0xf1/0x110 [drm_kms_helper]
[  147.924556]  drm_mode_cursor_universal+0x143/0x260 [drm]
[  147.924569]  drm_mode_cursor_common+0xc9/0x220 [drm]
[  147.924583]  ? drm_modeset_unlock+0x16/0x40 [drm]
[  147.924596]  ? drm_mode_setplane+0x2a0/0x2a0 [drm]
[  147.924608]  drm_mode_cursor_ioctl+0x4d/0x70 [drm]
[  147.924620]  drm_ioctl_kernel+0xac/0xf0 [drm]
[  147.924632]  drm_ioctl+0x201/0x3a0 [drm]
[  147.924643]  ? drm_mode_setplane+0x2a0/0x2a0 [drm]
[  147.924701]  amdgpu_drm_ioctl+0x49/0x80 [amdgpu]
[  147.924704]  do_vfs_ioctl+0xa4/0x630
[  147.924705]  ksys_ioctl+0x60/0x90
[  147.924707]  __x64_sys_ioctl+0x16/0x20
[  147.924709]  do_syscall_64+0x53/0x130
[  147.924711]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[  147.924712] RIP: 0033:0x7f12a7f24427
[  147.924713] Code: 00 00 90 48 8b 05 69 aa 0c 00 64 c7 00 26 00 00 00 48 c7 c0 ff ff ff ff c3 66 2e 0f 1f 84 00 00 00 00 00 b8 10 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 39 aa 0c 00 f7 d8 64 89 01 48
[  147.924714] RSP: 002b:00007ffc63baebd8 EFLAGS: 00003246 ORIG_RAX: 0000000000000010
[  147.924715] RAX: ffffffffffffffda RBX: 00005617e9dd6950 RCX: 00007f12a7f24427
[  147.924715] RDX: 00007ffc63baec10 RSI: 00000000c01c64a3 RDI: 000000000000000d
[  147.924716] RBP: 00007ffc63baec10 R08: 0000000000000001 R09: 0000000000003fff
[  147.924716] R10: 0000000000000780 R11: 0000000000003246 R12: 00000000c01c64a3
[  147.924717] R13: 000000000000000d R14: 00005617e82907c0 R15: 0000000000000279```

shmerl, 2019-09-06:
I also see such errors in the output of radeon-profile itself:

```DRM_IOCTL_AMDGPU_INFO: Invalid argument```

I opened a related bug for amdgpu itself: https://bugs.freedesktop.org/show_bug.cgi?id=111555

shmerl, 2019-11-07:
I don't get these errors anymore when using radeon-profile with kernel 5.4-rc6.

Oxalin, 2019-12-31:
This bug should be closed since it is not a radeon-profile issue. This is related to kernel/drm.

shmerl, 2019-12-31:
Agreed, it's a problem with powerplay.

closed 2019-12-31 by shmerl