marazmista/radeon-profile #159: Feature request: Create option to disable "Keep refreshing when hidden" by default

CypherNL, 2019-09-09:
**radeon-profile version:** 20190903

Currently, I'm using readeon-profile to get my GPU fans to work with a custom fan profile. I've set it up to apply the fan profile when the program starts up and minimize the program to the tray bar at boot. However it still refreshes the GPU info every second in the background and due to that radeon-profile uses 12% of CPU usage constantly.

Fortunately, there's an option in the right click menu of the tray icon to disable info refreshing in the background called "Keep refreshing when hidden" but the function is not persistent so every time i boot into my Desktop Environment i need to disable that function manually.

So my feature request is to have an option in the settings menu or in the right-click tray menu to disable "Keep refreshing when hidden" by default if it's possible.

parkerlreed, 2019-09-09:
Configuration > Check "Save selected fan control mode" and a save should make the daemon use that mode and then you can exit Radeon Profile.

CypherNL, 2019-09-09:
I've already have enabled that option but the fan curve is only active when having the main program open. I've enabled the Rades't show on my system monitor nor the fan profile has been applied. When i open the program, ieon-profile-daemon service in systemd and should start every time when i boot the computer. But it do hear my GPU fan spinning after a second or two.

I'm not sure how i get radeon-profile-daemon running independently from the GUI so only the daemon apply the fan curve. I've tried to set the daemon as data provider the the result is the same without it.

I've currently have these settings on.
[image]

Please note that the 60% fan speed is only on to confirm if the fan curve has been applied or not in the picture. The cure i use should be similar what the GPU does on Windows.

If you need more information. Please let me know.

marazmista, 2019-09-14:
I've commited the change (44381ef289ce61e8061980a86e7a43ba9597d097) where state of "Refresh when hidden" is saved in settings.

Daemon won't work independently from GUI because it doesn't have any steering logic. It is only used to put values into files that normal user has no write permissions.
There is a project which control fans from background, without GUI: https://github.com/grmat/amdgpu-fancontrol but I've never tried it.

CypherNL, 2019-09-22:
I've re-installed the AUR version recently and it doesn't refresh on boot anymore! Thanks!

closed 2019-09-22 by CypherNL