marazmista/radeon-profile #160: Translation of the application.

Ukol1, 2019-09-10:
Hi marazmista! I decided to translate your program into Russian. The program interface is translated, but the latest help is not included in the translation. When added to the translation code tr() in the uiEvents.cpp after compiling the help is not translated.
The tab is not an active State table, I also translated it.
`void radeon_profile::on_btn_connConfirmMethodInfo_clicked()
{
    QMessageBox::information(this, tr("Connection confirmation methods info"),
                             tr("0 - (Not recommended in multi user envirnoments) Disable conncection confirmation. Configured options in GUI may get stuck when user session is frozen \n\n" \
                             "1 - (Default) After some time of inactivity when GUI is connected, daemon sends request to GUI to check wheather it is alive. When response is not received it restores fan settings to system default\n\n" \
                             "2 - When no other commands has been sent to daemon, GUI sends periodically alive signals without deamon's requests. Select if you have issues with method 1"));
}` 
[image]
[image]

Ukol1, 2019-09-11:
Another conclusion about the diagonal is not properly displayed. I have two monitors, and how many monitors I have is not displayed correctly. What does this icon do, what function does it perform?

[image]

marazmista, 2019-09-14:
Thanks for translation! I added tr() to that message. (595dcd54445eecf12584e124a93e8abc86729cca)
This refresh button on right updates info in tabs GLX Info, Connectors and Module Info.

Ukol1, 2019-09-14:
Hi! I added uiEvents.cpp to the file tr(). Only the help title was translated, but the help text remained untranslated. Thanks for the answer! I can you straight for the file translation.

Ukol1, 2019-09-14:
Me strongly interested in why not active menu State table. I learned during the transfer that there is a power management. I have a video card OK food, long-term load appear distortion or turning off the graphics card. Fan controls don't help.

marazmista, 2019-09-15:
You can create pull request for translation if you want.
The State table is for overclocking and it is enabled when `pp_od_clk_voltage` file is available in sysfs.

Ukol1, 2019-09-15:
I do not have such a file in sys... The translation I did for Rosa Linux is included in the package. It is up to you to include the Russian translation in the program. Translation is custom.

marazmista, 2019-09-22:
I added translation 3a91e0a6df3fece5711dcf46fea67e02bdd03508
Thanks!

Ukol1, 2019-09-22:
Hello! It’s good that they made a change. I wanted to know how the pp_od_clk_voltage file is created? It turns out that all the program functions do not work for me.

Ukol1, 2019-09-24:
Hello! Thanks for adding the translation! I will close this topic, we have solved this problem. I’ll open another found more other problems ...

closed 2019-09-24 by Ukol1