marazmista/radeon-profile #161: Coil whine with Sapphire RX580 series

LiasOne, 2019-09-13:
When open radeon-profile starts coil whine and will end only after system restart.
Tested with 5 different RX580 Pulse and 2 RX580 nitro+.

Podesta, 2019-09-13:
Reported on #152

Mikasu-X14, 2019-09-19:
Try setting a fixed fan speed and switching to profiles and auto, play around with the fan profiles. It happened to me on an RX 580 Nitro+ and RX 590 Nitro+. I suppose it's just a bad setting for the fan, perhaps it gets too much or not enough voltage for it's current speed.
It's probably caused by the VBios and not by the software but still. I would get it to stop by switching through different fan settings and setting it back to my fan profile. Watch out, when they whine for too long, they may stop. Then, you have to let them cool down.

Mikasu-X14, 2020-02-26:
Faster way to get it to stop set a fixed fan speed, set the "mode" to "high" then changé your fans to your profile and switch back the mode to "auto". It should adjust the fan speed and may fix the whineing.

Oxalin, 2020-03-06:
@LiasOne is this still relevent?

mdeliu, 2020-05-05:
I can confirm this issue with my Radeon RX580 Nitro+, it also happens when I set the fan control profile to "auto". When I set the profile to "fixed", "custom curve" or reboot my machine it's gone. Also, the fan speed jumps between ~26% and ~48% every few seconds on idle.

- radeon-profile version 20200504
- daemon version 20190603
- updated amdgpu driver at Arch Linux

Actually not a big deal for me since I mostly use custom fan curves, but a bit annoying when I exit the application.

Mikasu-X14, 2020-05-06:
This is probably due to hardware issues rather than the program itself. Unless you tried with another fan control software that didn't cause this issue.

Oxalin, 2020-05-07:
I would be tempted to close the issue, this is not (99% sure) a bug related to radeon-profile. Radeon-profile only reveals the hardware "issue".

Podesta, 2020-05-07:
It is unlikely to be an issue directly on radeon-profile, but even more unlikely to be a hardware issue, considering this issue happens to almost all, if not all of the RX580, as demonstrated by the amount of issues that have been opened on this. There has been development done on radeon-profile to mitigate the issues caused by it, like fixing the hook, so running as sudo was no longer necessary, etc. The problem is with the underlying tool that it uses to control GPU, so it would be nice if someone with more knowledge could upstream the bug to the relevant place.

Mikasu-X14, 2020-05-07:
i think it's a vbios or hardware issue from sapphire. I have both an RX 580 and RX 590 that exhibits this behavior. It could just be with how amdgpu handles it when it reverts back to automatic settings. Also, the fans on special editions, which have blue LEDs, have already one issue where the LEDs will turn to a algea green over time and use, accelerated by high fan speeds and bearing temperatures. To be noted, since there are no alternative vbios available for most of these cards, we can't flash another to confirm it's a vbios issue. Only with the cards that have a vbios switch we can try with the silent vbios to see if it is indeed the source of the issue. To my knowledge, all Sapphire Nitro+ have two vbios that can be switched with a hardware switch while the system is turned off.

I will try for myself on my free time and give feed back on the silent bios' behavior with radeon-profile and amdgpu.

marazmista, 2020-05-11:
This can be verified manually without using radeon-profile:
1. Put fan control in manual mode: 
`echo "1" > /sys/class/drm/card0/device/hwmon/hwmon0/pwm1_enable`
2. Set custom fan speed (0-255), for example 50%:
`echo "128" > /sys/class/drm/card0/device/hwmon/hwmon0/pwm1`
3. Switch back to auto mode:
`echo "2" > /sys/class/drm/card0/device/hwmon/hwmon0/pwm1_enable`

This is basically all radeon-profile does in regard of fan speed control.

Oxalin, 2020-05-11:
What I meant by "Radeon-profile only reveals the hardware "issue"." was that the problem doesn't come from radeon-profile, but it is exposed by it, either that be a real hardware issue (a problematic or noisy coil on video card or in PSU), a vbios issue, a firmware issue or how the kernel driver is working.

While @marazmista suggests a way to test it, keep in mind this could also be related to power level.

Mikasu-X14, 2020-05-11:
i was thinking vbios because it would be plausible that the power to the fan for a given speed is incorrect or more likely, the drop or boost in fan speed when the card gets back to automatic would cause a too drastic change in voltage for the bearings, causing them to whine. That would be my guess.

jakbyte, 2020-05-23:
With auto I get a whine with the window open, and with the others I get it when the window is closed. I didn't get a whine until I started using custom curves.

jakbyte, 2020-05-23:
It seems that my custom curve settings aren't used when the window is closed despite having the daemon enabled.

AndreaMonzini, 2021-09-27:
I can confirm the same problem with RX 580, when i set "auto" i can hear a "Coil whine" noise.
I have to restart to solve the problem.
AMDGPU Mesa driver 21.2.1

JoneKone, 2021-09-28:
Coil whine is not an error nor and fault. It's symptom of 2 possibilities, 1. You bought a cheaply made GPU 2. You are running it on too high clock with no load. 3. both. So suddenly we find out that there is only 1 reason.  The power profile of the GPU hardware is made badly and we return to possibility 1. Radeon profiler is enabling a wrong power state and we return to possibility 2. But still Radeon profiler doesn't actually make your power states it's the bios (cheap) and user (doesn't know what he is doing)

So who's fault is it? GPU manufacturer's fault, every time. don't buy cheap GPU's. (I know the irony of this statement at time of writing)