marazmista/radeon-profile #162: Departure upon change

Ukol1, 2019-09-24:
Hello! I wanted to know how the pp_od_clk_voltage file is created? I tried to find him, but he is not. My State tab is not active.
After installing the latest update when changing, adding indicators on the top panel, the program crashes.
The ratio of the screen diagonal is not displayed correctly.
[image]
[image]
[image]

Ukol1, 2019-09-25:
Good! "Virtual screen", I logically called in translation "Monitor". There goes declension; "virtual screen", "virtual screens", "virtual screens". To check, I turned off the monitor and the inscription changed from "monitors" to " monitor". English does not convey an accurate translation from Russian. And then a virtual screen if there is a list of monitor outputs?!
The second monitor correctly displays the screen resolution and aspect ratio: 2560x1440 (16 : 9).
You wrote that the file responsible for manual tuning "pp_od_cl_voltage" is in sysfs... I have no such file in my system at all. State tab does not work.

marazmista, 2019-09-25:
> And then the virtual screen, if there is a list of outputs to the monitors...?!

It's just an information, you can have multiple X sessions etc.

> You wrote that the file responsible for manual tuning "pp_od_cl_voltage" is in sysfs... I have no such file in my system at all. State tab does not work.

So probably this functionality is not implemented in kernel for your hardware. If there is no file in sysfs, radeon-profile cannot control it and the tab is disabled.

Ukol1, 2019-09-26:
I have two systems, on Manjaro there is a 5.3 core, on Rosa 5.2, there is no manual adjustment in two systems.

closed 2020-05-11 by marazmista