marazmista/radeon-profile #163: vega64 does not apply memory / core clock

MartB, 2019-09-29:
The memory clock in the states table is not applied to the device, for some reason nothing happens when the memory clocks get changed.

Running the following command manually works though.
`sudo echo "m 3 1050 1000" > "/sys/class/drm/card0/device/pp_od_clk_voltage"`

Core clock changes seem to be broken aswell.

Can someone point me into the right direction?

Oxalin, 2020-03-06:
Is this still relevent?

Benzhaomin, 2020-05-27:
Running with sudo radeon-profile still shows

```
QIODevice::write (QLocalSocket): device not open
Failed sending signal:  "2#s 0 852 800#/sys/class/drm/card0/device/pp_od_clk_voltage#2#s 1 991 900#/sys/class/drm/card0/device/pp_od_clk_voltage#2#s 2 1084 950#/sys/class/drm/card0/device/pp_od_clk_voltage#2#s 3 1138 1000#/sys/class/drm/card0/device/pp_od_clk_voltage#2#s 4 1200 1010#/sys/class/drm/card0/device/pp_od_clk_voltage#2#s 5 1401 1010#/sys/class/drm/card0/device/pp_od_clk_voltage#2#s 6 1536 1030#/sys/class/drm/card0/device/pp_od_clk_voltage#2#s 7 1580 1040#/sys/class/drm/card0/device/pp_od_clk_voltage#"
```

marazmista, 2020-06-09:
@Benzhaomin @MartB
 I pushed a fix in commit 65318da2d39b5e6a153e3d74e6810c4d63de8435. Please try it, I think it should work fine now.

Benzhaomin, 2020-06-10:
Looks fixed to me, thanks.

JoneKone, 2020-08-30:
I'm also currently experiencing this issue, I installed a version "Git unstable" (git develop repository) from PPA Ubuntu 20.04 Vega 56, It will change the Power limit, but won't touch pp_od_clk_voltage at all. I tested and CoreControl works fine. You will ask me what version is on, but currently when Corecontrol is running it won't allow me to start Radeon-Profile it is telling me to "At least one core state has to be enabled"

i quest it won't allow me to start it at the same time as CoreCtrl, nor after it was started and stopped. 

On the picture it says v20200824

[image]
[image]

Razer0123, 2020-10-02:
Not working for me either

Nostromo-KM, 2021-04-08:
Fresh install - Manjaro with Cinnamon UI
Kernel 5.4
- Installed from AUR
- Deamon installed from AUR
- Deamon configed to auto-run
- GRUB amdgpu.ppfeaturemask=0xffffffff amdgpu.dpm=1

Created new frequency/voltage states tables but cannot apply them
Tried running sudo - no luck
Tried to do the same with Core-Ctrl - it works.

closed 2024-11-11 by MartB