marazmista/radeon-profile #164: console error log message (repeats continuously)

splace, 2019-10-06:
```` 
DRM_IOCTL_AMDGPU_INFO: Invalid argument
````

splace, 2019-10-06:
update 20190903
(may have always done this, only now checked console.)

eiglow, 2019-10-22:
I get this too, it repeats every refresh.

eiglow, 2019-10-30:
@splace What does `sensors -u` display for you? For my GPU, I get the following:
```amdgpu-pci-1c00
Adapter: PCI adapter
vddgfx:
ERROR: Can't get value of subfeature in0_input: Can't read
fan1:
  fan1_input: 0.000
temp1:
  temp1_input: 45.000
  temp1_crit: 120.000
  temp1_crit_hyst: 90.000
power1:
ERROR: Can't get value of subfeature power1_average: Can't read
ERROR: Can't get value of subfeature power1_cap: Can't read
```

Perhaps these errors are related to this issue?

splace, 2019-10-30:
i get the same, not sure about being related though.

eiglow, 2019-10-30:
What GPU do you have? I had the same issue with both an r5 240 and HD 7970.

splace, 2019-10-30:
Radeon R7 250

marazmista, 2019-11-11:
This is related to getting info from kernel via ioctls. I don't know why this happens though, I checked arguments in kernel and they seem to be fine.
This also may be the reason why GPU usage doesn't work for some systems.

Oxalin, 2020-03-06:
@splace is this still relevent? If so, could you tell us more about your setup? Your error points to a kernel problem. Which version are you using?

splace, 2020-03-06:
```
>uname -a
Linux solus-ssd 5.5.6-149.current #1 SMP PREEMPT Wed Feb 26 03:52:34 UTC 2020 x86_64 GNU/Linux
```

as with @eiglow  i use the default amdgpu driver module.

this has been the same through ~30+ kernel updates (and probably always on solus ~100+ 3 years) and something like the same number of updates/versions of the amd driver.

since originally reporting i have reinstalled the OS without change to this AFAIK.

does solus compile the amd driver with something optional, you use, not selected?

from the name i guess its a call to retrieve some 'live' info from the driver, but profile seems to have all data correctly.

```

>radeon-profile

(radeon-profile:2941): Gtk-WARNING **: 15:23:35.679: Unable to locate theme engine in module_path: "murrine",

(radeon-profile:2941): Gtk-WARNING **: 15:23:35.680: Unable to locate theme engine in module_path: "murrine",

(radeon-profile:2941): Gtk-WARNING **: 15:23:35.681: Unable to locate theme engine in module_path: "murrine",
DRM_IOCTL_AMDGPU_INFO: Invalid argument
DRM_IOCTL_AMDGPU_INFO: Invalid argument
DRM_IOCTL_AMDGPU_INFO: Invalid argument
DRM_IOCTL_AMDGPU_INFO: Invalid argument
DRM_IOCTL_AMDGPU_INFO: Invalid argument
DRM_IOCTL_AMDGPU_INFO: Invalid argument
DRM_IOCTL_AMDGPU_INFO: Invalid argument
Segmentation fault

```

seg-fault occurs when shutting window, guess is unrelated.

Oxalin, 2020-03-08:
@splace I could create a patch if you are willing to test it. It will help us to identify what is wrong.
@eiglow Are you using the same distro as the one splace is using?

Oxalin, 2020-03-09:
This link may be relevent to the error message you are seeing: https://www.mail-archive.com/amd-gfx@lists.freedesktop.org/msg38150.html
If so, we should have a way to better identify which value is invalid and we shouldn't flood the log,

eiglow, 2020-03-09:
I'm using a different distro, Fedora, but Fedora always has a pretty vanilla kernel as far as I know. I'm willing to test a patch, I'll do it with both my r5 240 and HD 7970 if you want.

Oxalin, 2020-03-11:
@eiglow @splace Could you compile my master branch to see what you get when launching radeon-profile from a console?
https://github.com/Oxalin/radeon-profile/commit/4621199d010dbefb63605436936b84601d8a9afe

Oxalin, 2020-03-22:
@eiglow @splace Please, replace previous suggestion by this one which decodes the sensor code: https://github.com/Oxalin/radeon-profile/commit/ed82d2b78b9dea06d5e8fa3a6c2821e65a79fa83

closed 2020-05-04 by marazmista

eiglow, 2020-09-06:
@Oxalin Finally got around to this. I compiled and ran your fork, but I still got the same error message (`DRM_IOCTL_AMDGPU_INFO: Invalid argument`) in the terminal.

splace, 2020-11-18:
having to run like this....

> radeon-profile > /dev/null 2>&1

since its completely flooding the log.

splace, 2020-11-18:
issue closed due to a patch that is known NOT to fix it!

doesn't this just hide any help the information here might be to someone else and invite duplicates?

splace, 2021-04-17:
reopen?