marazmista/radeon-profile #165: Can'r run radeon-profile as root UI does not show, tray icon unresponsive

Dantali0n, 2019-10-08:
Hello I do not want to use radeon-daemon but can't start radeon-profile as root anymore. When started as root the UI does not show and the tray icon is frozen.

When build with debug no valuable information indicating any problem is put into the console.

I use Ubuntu 18.04 with qt5.13

hydrian, 2019-12-06:
You don't run the radeon-profile as root. You run the radeon-profile-daemon as root. Then radeon-profile connects the radeon-profile-daemon as a client. This way, regular users don't need root access to make GPU settings changes which often require root access.

Dantali0n, 2019-12-08:
I don't want to use the daemon, the program used to work by running radeon-profile as root and you could run it without using the daemon at all.

Does this mean this functionality has been removed?

hydrian, 2019-12-08:
I've never seen it work without the deamon and got my RX570 during crypto mining scarcity which is when I started using low-profile(-daemon).

Also, from a security point of view, what your doing is bad idea. Some distros may be blocking it with policykit. 

Also, you system tray is running as you logged in user, not root. That's probably why the icon is unresponsive.

hydrian, 2019-12-11:
Oh. I forgot about this method. You have to setup sudo to allow you to run radeon-profile. 

Right from the documentation (https://github.com/marazmista/radeon-profile/blob/master/README.md): 
Otherwise, the app needs to be run with root privileges for changing power profiles (and clocks readings sometimes). You can add
`username ALL = NOPASSWD: /usr/bin/radeon-profile` 
to your /etc/sudoers file. 

Here is tip for run app as normal user but involves change permissions to system files: http://bit.ly/1dvQMhS

hydrian, 2019-12-11:
Actually, it is better to make add that sudo line to /etc/sudoers.d/radeon-profile. That way the change will survive OS upgrades which was probably how it became broken on your system.

Dantali0n, 2019-12-11:
> Oh. I forgot about this method. You have to setup sudo to allow you to run radeon-profile.
>
> Right from the documentation (https://github.com/marazmista/radeon-profile/blob/master/README.md): [...]

" You **have** to setup sudo to allow you to run radeon-profile".
* The only difference adding this line to sudoers makes is that it will no longer require to start the application by typing the sudo password. The application still runs as root and it makes no difference in the permissions it has or how it runs.

"That way the change will survive OS upgrades which was probably how it became broken on your system."
* There is no reason to believe that this point that this is the case, actually I haven't even installed the program I just run it from the source directory directly.

The documentation clearly states that it is possible to run the application this way:
"Install and run radeon-profile-daemon (https://github.com/marazmista/radeon-profile-daemon) for using this app as normal user. **Otherwise app need to be run with root privilages for changing power profiles** (and clocks readings sometimes)".

This issue is valid for these reasons; it used to work and the documentation states that it should work. Telling me that I shouldn't run it as root and how It's insecure does not contribute towards solving it not working.

hydrian, 2019-12-11:
Okay, first you want to say you want to run it without the daemon. This means you have to run it as root via sudo. 

I also verified you can run radeon-profile with sudo without any issue with the system tray with current master (pulled today) of radeon-profile without the daemon. It sounds more like an environmental issue. 

I'm running...
Mint 19.0 with kernel 5.0.0-37 
radeon-profile 20190903 /w Qt 5.9.4

Oxalin, 2020-03-12:
@Dantali0n is this still a problem?

Dantali0n, 2020-03-12:
> @Dantali0n is this still a problem?

Yes, this is still a problem on Ubuntu 18.04 with qt-5.11, on my Arch installation it runs fine however, that machine does not have an AMD graphics card.