marazmista/radeon-profile #167: Can't compile on Ubuntu 19.10

bkdwt, 2019-10-12:
When I try to compile with qmake, this error appears:
´Project MESSAGE: Building for release
Project ERROR: Unknown module(s) in QT: charts`

I have checked '/usr/lib/x86_64-linux-gnu/qtchooser/qt5.conf' and the file contain this:

'/usr/lib/qt5/bin
/usr/lib/x86_64-linux-gnu'

Check 'qtchooser -l':

'qtchooser -l
4
5
default
qt4-x86_64-linux-gnu
qt4
qt5-x86_64-linux-gnu
qt5'

Whatever option I choose with 'qmake -qt doesn't work.

trytip, 2019-10-24:
install these should work after with qmake
`apt install libqt5charts5 libqt5charts5-dev`

bkdwt, 2019-11-05:
> install these should work after with qmake
> `apt install libqt5charts5 libqt5charts5-dev`

Thanks! Works. ;)

closed 2019-11-05 by bkdwt

mikelward, 2020-05-24:
Could you update https://github.com/marazmista/radeon-profile/blob/master/README.md to reflect this please?

marazmista, 2020-06-03:
@mikelward Updated (34cef06d0643c77d79239c304624417fc5036911)