marazmista/radeon-profile #168: Manual Frequencies not longer fine tunable ?

MasterCATZ, 2019-10-26:
just built 20190903 

but now it only gives me a few options to select from for frequencies instead of putting my own in 

I normally run GPU idle @ 300 and most games run fine 20-30 FPS  , and running on Performance mode is overkill 60 - 200 FPS  and 90 deg heat @ 100% fan ,  main issue is I want to lower memory speed just a tad for stability 

[image]

MasterCATZ, 2019-10-28:
is amdgpu.dpm=1 meant to be on or off for radeon-profile to work ?
using Radeon=Profile 20190903 / 5.3.6-050306-generic / 3.0 Mesa 19.3.0-devel (git-3a20ef4 2019-10-22 bionic-oibaf-ppa)

if I have profiles set manually I can adjust the GPU clocks , but it completely ignores fan speeds and keeps running low fan speeds , I can not get it to hold 50% plus and it keeps dropping to 20% roasting it @ 100 deg 200watts
( I know this card peaks out at 330 watts power draw R9 290 from crypto mining days and using performance bios ),
My gaming fan profile is normally 1:1 , ie 70 deg = 70%

if I set it to profile_peak ( with a low core speed set ie 300mhz ) it will run core 483mhz with 60% fan speed 90 watts
if I have it set to a high core speed ie 950mhz ) it will run that core speed with passable fan speeds
( anything below that default max core speed runs 483mhz
thankfully this still at least gives me 60 FPS in Doom3 so its what I am currently using when playing games finally Ubuntu / Steam is running games sweet with Vulkan

so I just throw it on low profile when using desktop but it runs high temps @ 300mhz because fan speeds are lower than I would like

I think it might be because of power cap settings which is set at 208w but I have no idea how to change this

https://drive.google.com/drive/folde...bs?usp=sharing

thats a shared folder with screen shots / screen recording etc , apparently I have no attachment permissions to add them here

with Vulkan now supporting Multi GPU i really want to get radeon-profile working so i can throw my numerous R9 290's into a PC again

I also would like a Underclocking option , their is a nice sweet spot performance / power usage for multi GPU

MasterCATZ, 2019-11-01:
ended up editing bios for my frequencies I wanted 
removing amdgpu.dpm=1  allowed me to control the fans without something randomly forcing fans to 25% regardless of temp , and I know its not the bios because that's set on 80% and something was even forcing that to 25% 
auto freq control was lost without dpm it just sticks on high core low memory instead of using freq in between based on load standard profile seems to be mid range setting at least

marazmista, 2019-11-11:
> but now it only gives me a few options to select from for frequencies instead of putting my own in

Frequencies were never fine tunable. Before checkbox list there was slider that had same points as this list.

> I also would like a Underclocking option , their is a nice sweet spot performance / power usage for multi GPU

Underclocking in percents? What happen if you try manually put negative percent values to appropriate sysfs files? Would it work?

GM-Script-Writer-62850, 2020-03-30:
I think what you you are asking for is in the states table, you need to enable OC mode to access it, boot parameter for me is `amdgpu.ppfeaturemask=0xffffffff`
[image]