marazmista/radeon-profile #169: Feature Request: FreeSync / VRR

D33M0N, 2019-10-29:
Could it be possible to show if FreeSync / VRR is currently enabled and active? And if yes, then on which display and process/program/game is controlling it?

ripper81, 2019-11-01:
Yes we need a complete gui tool for radeon GPU's on linux. It would be really good to have functions like screen resolution, refreshrate, Vsync, Freesync and screensetup integrated into radeon-profile. Radeon-Profile could easily be a complete desktopindependent radeon settings tool.

Mikasu-X14, 2020-02-26:
You can read it as some raw data in at the bottom of the first tab. Under the display informations. It will be called "tearfree" and "vrr", it's either 1 or 0. True or False. If both are true, you can safely assume FreeSync is enable and functionning correctly. If it's not, try DisplayPort instead if you were on HDMI.

ripper81, 2020-02-26:
I am aware of those information. What i like to see in radeon-profile is a gui switch to activate these options in an easy way.

Mikasu-X14, 2020-02-26:
I don't know why you would like to disable them but they need a reboot in order to be applied, correct me if i'm wrong. To set vrr_capable and TearFree, you need to set them in a config file for xorg (/etc/X11/xorg.conf.d/)

parkerlreed, 2020-02-26:
HDMI isn't supported for VRR. Only works in their prop Windows code.

ripper81, 2020-02-29:
VRR (Freesync) and TearFree (Pageflipping) are not enabled by default. At least not on Kubuntu 19.10. You have to set the options by editing the file "10-amdgpu.conf" in /usr/share/X11/xorg.conf.d/

The file should look like this after adding the needed options:

Section "OutputClass"
	Identifier "AMDgpu"
	MatchDriver "amdgpu"
	Driver "amdgpu"
	Option "VariableRefresh" "true"
	Option "TearFree" "true"
EndSection

Mikasu-X14, 2020-03-02:
vrr_capable = on and tearfree = on should be the correct settings.

D33M0N, 2020-03-02:
This "tearfree" is vsync or something else?
Anyway. What I originally wanted to see, was if there is a way to make sure, that some program / display is right now currently using freesync/vrr.

Yes, knowing if my setup and settings are correct also (or possibly helped to setup correctly with radeon-profile) is helpful also as it's like pre-requisite for the thing to actually work and being used, but I was asking for the next step -- actual confirmation/info that some game or demo or something is actually controlling it right now on some display. Just having tearfree = on and vrr_capable = 1 doesn't mean much in that regard... it just means my computer is ready and willing..... but ... is anything actually taking advantage of it, that is the question!

Mikasu-X14, 2020-03-02:
it's not something that games can actualy control. if it's enabled, it's working. the goal of tearfree is, yes, quite similar to vsync. vrr_capable is to vary in refresh rate so (in fullscreen applications and with compositor disabled) the refresh rate will vary depending on the number of frames that were drawn.

D33M0N, 2020-03-07:
not according to this: https://www.amd.com/en/support/kb/faq/gpu-754
There is a (probably outdated at this point) list of applications that "support it". Apparently the application has somehow to turn it "on" or "use it". Maybe it's done driver side or something... in vulkan or dxvk or opengl or whatever, but it's sadly NOT "working" as you say, for everything if it's turned on. It may not be working for the current application I expect it to be working and I want to know about it. Also I want to know if it is actually working.

Mikasu-X14, 2020-03-07:
it's not application controlled. it's controlled by the driver or xorg. I can say from experience that it seems to work with pretty much every game. I can't give you any examples where it did not work. Perhaps the list of compatible applications is applications they tested Freesync with to ensure that it's working and they cannot test every software in the world. I will say that, for myself, even games running on wine/proton seemed like they were working with freesync. There's only osu!lazer that would have a very light amount of screen tearing, almost unnoticeable.

For the two applications thing, if they both go fullscreen, it's the one you're actively on i'd say. Otherwise, if in windowed, i do not know.

Mikasu-X14, 2020-03-07:
To test for yourself, maybe connect your monitor over hdmi, to be sure freesync will not work, play some games then connect your monitor over DisplayPort and test the same games and see if there's a difference.

Mikasu-X14, 2020-05-06:
Clarification, VRR and TearFree should apply whenever your desktop compositor is disabled windowed or fullscreen. Depending on your DE, the way to disable the compositor will probably be different. Though, many games disable desktop compositors by default.