marazmista/radeon-profile #170: Can't bring back application from tray icon in Ubuntu 19.04

hatsunearu, 2019-11-03:
Hi, I'm having trouble finding the procedure to restore the application after it has been minimized to tray in Ubuntu 19.04, with the default DE (Gnome 3).

chipshopness, 2019-11-06:
Same here. There would need to be a 'Show radeon-profile' item on the menu maybe. Are you using the KStatusNotifierItem/AppIndicator  gnome-shell extension? The [readme](https://github.com/ubuntu/gnome-shell-extension-appindicator) says that doubleclicking the tray icon brings up the main window if implemented by the application but I've yet to use anything that does this.

hatsunearu, 2019-11-06:
I am not; is that a valid workaround?
This makes the app unusable because suspend-resume makes the fan control unresponsive :(