marazmista/radeon-profile #171: Issue: Not switch to Radeon card with DRI_PRIME=1

Roker2, 2019-11-05:
Hi! My laptp is Dell Inspiron 15 3576 with Radeon 520 Mobile. I try `sudo env DRI_PRIME=1 radeon-profile` but I see settings for Intel videocard.
Version: 20190903

More info:
sudo env DRI_PRIME=1 lspci -v -s 01:00.0
01:00.0 Display controller: Advanced Micro Devices, Inc. [AMD/ATI] Jet PRO [Radeon R5 M230 / R7 M260DX / Radeon 520 Mobile] (rev c3)
	Subsystem: Dell Jet PRO [Radeon R5 M230 / R7 M260DX / Radeon 520 Mobile]
	Flags: bus master, fast devsel, latency 0, IRQ 131
	Memory at c0000000 (64-bit, prefetchable) [size=256M]
	Memory at d0000000 (64-bit, non-prefetchable) [size=256K]
	I/O ports at e000 [size=256]
	Expansion ROM at d0040000 [disabled] [size=128K]
	Capabilities: [48] Vendor Specific Information: Len=08 <?>
	Capabilities: [50] Power Management version 3
	Capabilities: [58] Express Legacy Endpoint, MSI 00
	Capabilities: [a0] MSI: Enable+ Count=1/1 Maskable- 64bit+
	Capabilities: [100] Vendor Specific Information: ID=0001 Rev=1 Len=010 <?>
	Capabilities: [150] Advanced Error Reporting
	Capabilities: [270] Secondary PCI Express <?>
	Kernel driver in use: radeon
	Kernel modules: radeon, amdgpu

marazmista, 2019-11-11:
I suppose that in `/sys/class/drm/` you have two cards, (`card0`,` card1` maybe?). Could you paste contents of file named `uevent` from `device` directory inside each card? Something like this: `/sys/class/drm/card0/device/uevent`. Might be something wrong with card detection.

Roker2, 2019-11-15:
`ls /sys/class/drm/
card0  card1  card1-HDMI-A-1  card1-eDP-1  renderD128  renderD129  ttm  version`
`DRIVER=radeon
PCI_CLASS=38000
PCI_ID=1002:6665
PCI_SUBSYS_ID=1028:083F
PCI_SLOT_NAME=0000:01:00.0
MODALIAS=pci:v00001002d00006665sv00001028sd0000083Fbc03sc80i00`

Roker2, 2019-11-24:
So, when I open program, I see this
`sudo radeon-profile
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
Translation not found.
QFSFileEngine::open: No file name specified
QFSFileEngine::open: No file name specified
QFSFileEngine::open: No file name specified
QFSFileEngine::open: No file name specified
QFSFileEngine::open: No file name specified
QFSFileEngine::open: No file name specified`

Roker2, 2019-11-24:
Oh, lol
I tried `sudo env DRI_PRIME=1 radeon-profile` and got this
`sudo env DRI_PRIME=1 radeon-profile
[sudo] password for dmitry: 
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
Translation not found.
/dev/dri/renderD128: Invalid argument
/dev/dri/card0: Invalid argument
QFSFileEngine::open: No file name specified
QFSFileEngine::open: No file name specified
DRM_IOCTL_RADEON_INFO: Bad file descriptor
DRM_IOCTL_RADEON_INFO: Bad file descriptor`