marazmista/radeon-profile #172: When exit radeon profile fans are put on auto and can not unlock

MasterCATZ, 2019-11-06:
so it  seems me losing access to fan control happens when radeon profile is exited and then switches the card to Auto , for some reason after this its permanently on #2 and will not allow anything to change it back to manual , AMD's fan settings being ultra low does not help either ...

marazmista, 2019-11-11:
Strange. Does this happen when you modify sysfs files manually?

ruvaldak, 2019-11-24:
I think I've got this same issue. On Manjaro myself. Can change the profile in app and it works fine, but as soon as I exit it switches back to Auto. I know this because it causes a very loud and audible high pitched whine as soon as it goes to Auto. Reopening radeon profile fixes it.