marazmista/radeon-profile #175: Incorrect GPU usage

ruvaldak, 2019-11-24:
GPU usage seems to almost always be at 0%. occasionally it'll say like 5% or something, but it's usually at 0%, even while playing a heavily demanding game. radeontop seems to provide, at least, a more believable gpu usage.

Myrddin-Wyllt, 2020-01-25:
Confirmed on my RX 480 with Manjaro that this still is the case with the current master branch. This program & daemon combo is the best solution I've found for managing GPU profiles for AMD graphics on Linux.

Oxalin, 2020-03-03:
Is this still relevant? Please see issue #70.

ruvaldak, 2020-03-03:
[image]

[image]

Two images taken not long after another, showing the extreme discrepancy of gpu usage compared to radeontop, which I've found to show relatively seemingly accurate gpu usage. These were taken today, after installing the latest from the AUR including the daemon (which is running)

[image]

Oxalin, 2020-03-05:
While you have both RadeonTop and Radeon-profile running, could you check the value in gpu_busy_percent. Just go under /sys/class/drm/card0/device and "cat gpu_busy_percent" at repetition. That way, we will see if the problem comes from Radeon-profile or from the value reported by the amdgpu module.

From what I can see on my system, Radeon-profile and the value in gpu_busy_percent are very similar. However, RadeonTop is giving me a different value that doesn't fit with gpu_busy_percent.

ruvaldak, 2020-03-08:
Mine is the same. Trying to capture gpu_busy_percent and Radeon-profile at the same value is difficult but `watch -n 0.1 cat /sys/class/drm/card0/device/cat gpu_busy_percent` is relatively in line with what Radeon-profile displays, and neither show anything similar to what Radeontop reports. However, gpu_busy_percent, when reporting faster like that it makes more sense. I figure doing something like gathering gpu_busy_percent over x interval and averaging it might provide a similar value to Radeontop.

Oxalin, 2020-03-11:
A few notes:
- Radeontop creates 120 samples / sec. You could change the first sleep for "sleep 0.0083"
- After testing your code, my result is closer to radeon-profile than to radeontop. I would have to look at radeontop's code.

ruvaldak, 2020-03-11:
120 samples/second? Ill give that a looksie, see how close it is to Radeontop on my system.

On radeon-profile, for my system, it never seems to report an average usage, more that it just grabs whatever value is in gpu_busy_percent every second. Now, this is just a logical guess, but from what Im seeing, it seems like gpu_busy_percent is almost *too* real-time to grab from once a second. Something like gathering the tiniest fluctuations in usage (so at any moment the gpu being i.e. 0% busy rendering something that one second you grabbed it while in reality you dont actually have 100% of your gpu free to utilize because the next second it jumps it to like 100%). So radeon-profile reports tend to be something like 0% one second, 100% the next, 20% the next, and so on - random values that dont really add up to much. Im curious as to how yours is, and whether or not its a problem with my system only.

Oxalin, 2020-03-23:
So we could increase the sample rate in radeon-profile.

ruvaldak, 2020-03-28:
I might try to rewrite/fix the getGpuUsage in ioctHandler, submit a PR, sometime in the near future. Have school to focus on atm, and I gotta figure out how exactly Id want to go about it. Unless someone gets to it before me, in which case radical. Does look like, whatever its doing, its doing it with a frequency of 150, though I dont quite understand how the current one works, I cant really see how it gets gpu usage at all.