marazmista/radeon-profile #177: No longer able to modify fan speeds on Ubuntu 18.04

evolze, 2019-12-04:
Hello,

I'm relatively new to both the Linux and GitHub communities. I experienced an issue with the version of this app that was recently updated (via Launchpad (PPA)) on 12/02/2019. 

I was updating my packages (via the PPA in Launchpad), and once I restarted my system, the Radeon Profile application no longer allows me to set a custom fan speed. I've tried everything on my end, such as deleting the PPA, removing the application with all of its dependencies, reinstalling, etc.

I've provided screenshots to show the issue that I'm having. 
[image]
[image]

Any additional help would be much appreciated. Thank you for taking a look at this.

thunder1410, 2019-12-04:
I am also having this issues. The PPA has a quite outdated version of radeon profile. It doesn't get new versions because the auto compile files, because the code doesn't follow the debian build files. For some reason it managed to compile the deamon but not the main program, which pushed a update for the deamon without updating the main program. I am also searching for a solution (doesn't want to compile from source).

evolze, 2019-12-04:
> I am also having this issues. The PPA has a quite outdated version of radeon profile. It doesn't get new versions because the auto compile files, because the code doesn't follow the debian build [...]

I'm in the same boat as you. I'm unable to compile it from the source too! I've tried for hours, but have gotten nowhere. Hopefully, this will be fixed soon!

aaron576, 2019-12-05:
i had been using the ppa as well and had this issue. one solution is to run radeon profile as root but thats not very convenient. i was able to compile from source on a fresh install of 18.04 though. make sure you install 

libxrandr-dev

qt5-default (this is a typo on in the instructions. its **default** not default**_s_** that you want)

libqt5charts-dev

and it should compile fine.

MasterCATZ, 2019-12-06:
its not  radeon-profile fault 
( apart from how he has it go into Auto when program closes )

 it is amdgpu / hwmon / powerplay  

/sys/class/drm/card1/device/hwmon/hwmon1/pwm1_enable

is stuck on #2 and its been pissing me off since kernel 5

I have been having to recompile every kernel so far just to disable it  

linux kernel change true to false 
drivers/gpu/drm/amd/powerplay/hwmgr/hwmgr.c 

hwmgr->fan_ctrl_is_in_default_mode = false;

and not let anything make pwm1_enable #2 / Auto  or else it will get stuck their again

thunder1410, 2019-12-06:
The ppa is fixed 2 hours ago! It updated to the latest version.

evolze, 2019-12-19:
Sorry, it took me so long to respond. I was able to get it working on Ubuntu again! Thanks for all of your help!

closed 2019-12-19 by evolze

muzena, 2019-12-23:
@evolze 
Which papa do you use?

evolze, 2019-12-24:
@muzena I used the ppa:trebelnik-stefina/radeon-profile

Here's the exact link on [Launchpad](https://launchpad.net/~trebelnik-stefina/+archive/ubuntu/radeon-profile).

muzena, 2019-12-24:
> @muzena I used the ppa:trebelnik-stefina/radeon-profile
> 
> Here's the exact link on [Launchpad](https://launchpad.net/~trebelnik-stefina/+archive/ubuntu/radeon-profile).

No this is testing repo.
This is Radeon profile ppa stanke and git:
https://launchpad.net/~radeon-profile

evolze, 2019-12-24:
My apologies @muzena 

I was using the incorrect repo this whole time. Thank you for your help!