marazmista/radeon-profile #178: CLI interface

hydrian, 2019-12-08:
Is there a CLI interface for switching profiles? I'd like to script profile changes depending on what I'm doing. This way i don't have to hear the GPU fan hum all the time.

sinisa94, 2019-12-09:
Read documetation

hydrian, 2019-12-09:
What documentation? All I see is https://github.com/marazmista/radeon-profile/blob/master/README.md and that talks nothing about CLI arguments. Is there someplace else the documentation is hiding and I'm not find it.

marazmista, 2020-03-27:
No, I don't have plans regarding CLI interface, but puhsing values directly to sysfs files could be treated sort of like console interface.

closed 2020-03-27 by marazmista

daroczig, 2020-06-30:
As far as I know, `sysfs` updates requires root-level access, while it would be much more convenient to rely on `radeon-profile-daemon` as a user -- if it's already running, so I still hope there's some chance for this in the future, although I understand it's not a priority.

hydrian, 2020-07-02:
@daroczig Exactly. That's why I asked if radeon-profile had a CLI interface. Also, you can make a lot of changes with radeon-profile's profiles. Having a switch that could just switch from pre-saved profiles would be a great enhancement.

hydrian, 2020-07-02:
Basically, I want to be able to set my fan curve profiles and performance mode when I start up a game automatically(read scripted) and revert it back once I'm done playing the game.