marazmista/radeon-profile #184: Not Working Without Sudo Even WITH radeon-profile-daemon systemd service running

gardotd426, 2020-02-12:
I've installed this many times before, and I know for a fact it works with this card. However I just tried to install it on a new install of Manjaro, and it doesn't work, there's something wrong with the daemon. When I run it without sudo, the Window has no information in it (because the permissions aren't correct), but when I run `sudo radeon-profile` of course it works. But I've already enabled the service, and rebooted, and double-checked that `radeon-profile-daemon.service` was active and running with `systemctl status radeon-profile-daemon.service`. The output:
```
radeon-profile-daemon.service - radeon-profi>
   Loaded: loaded (/usr/lib/systemd/system/rad>
   Active: active (running) since Wed 2020-02->
 Main PID: 27784 (radeon-profile-)
    Tasks: 2 (limit: 4915)
   Memory: 1.4M
   CGroup: /system.slice/radeon-profile-daemon>
           └─27784 /usr/bin/radeon-profile-dae>

Feb 12 10:56:06 manjaro systemd[1]: Started ra>
Feb 12 10:56:06 manjaro radeon-profile-daemon[>
Feb 12 10:56:06 manjaro radeon-profile-daemon[>
lines 1-12/12 (END)...skipping...
● radeon-profile-daemon.service - radeon-profile daemon
   Loaded: loaded (/usr/lib/systemd/system/radeon-profile-daemon.service; enabled; vendor preset: disabled)
   Active: active (running) since Wed 2020-02-12 10:56:06 EST; 43s ago
 Main PID: 27784 (radeon-profile-)
    Tasks: 2 (limit: 4915)
   Memory: 1.4M
   CGroup: /system.slice/radeon-profile-daemon.service
           └─27784 /usr/bin/radeon-profile-daemon

Feb 12 10:56:06 manjaro systemd[1]: Started radeon-profile daemon.
Feb 12 10:56:06 manjaro radeon-profile-daemon[27784]: "radeon-profile-daemon (v. 20190603)"
Feb 12 10:56:06 manjaro radeon-profile-daemon[27784]: Awaiting connections...
```
 I've tried rebooting, reinstalling, using the AUR package, building it myself from the github repo, everything. It absolutely will not work.

Mikasu-X14, 2020-02-26:
From my experience on Manjaro, the AUR package should work just fine. Maybe try to see if radeon-profile is launching before it's daemon. In other words, quit radeon-profile and relaunch it and see if it works. It would be important to check your settings in radeon-profile regarding the daemon too.
At least, it looks like the daemon is working correctly.

bassamanator, 2020-05-12:
@Mikasu-X14 Where to check the settings regarding the daemon in radeon-profile?

gardotd426, 2020-05-12:
There are no daemon settings, it's just a daemon. It runs, or doesn't run.
The only daemon-related settings are two configuration options in the
regular radeon-profile configuration menu in the GUI, and those are "Use
daemon as a data provider," which no idea what the hell that means, and
then "Refresh data without request." There are no other options to for the
daemon itself.

bassamanator, 2020-05-13:
Hmm well I thought that would solve my problem. I'm getting a bad unit file settings when I try to start the daemon.

```sudo systemctl start radeon-profile-daemon.service
Failed to start radeon-profile-daemon.service: Unit radeon-profile-daemon.service has a bad unit file setting.
See system logs and 'systemctl status radeon-profile-daemon.service' for details.```