marazmista/radeon-profile #186: Mode returns to "Performance" a second after pressing "Battery" or "Balanced" button

Nowaker, 2020-02-13:
And I don't know what that means.

nikerrr, 2020-02-14:
I think this behaviour related to : [power_dpm_state](https://dri.freedesktop.org/docs/drm/gpu/amdgpu.html#power-dpm-state). In short: "The power_dpm_state file is a legacy interface and is only provided for backwards compatibility. On newer GPU's selecting battery/balanced/performance does nothing".

MangaD, 2020-03-16:
I have the same problem. Doing:

```sh
echo profile > /sys/class/drm/card0/device/power_method
```

changes nothing either.

If this is a legacy interface, how can we change the power method nowadays?

Thank you.

Akrai, 2020-05-06:
I think those buttons would do the same as controlling the frequency of the card, so you can use the dropdown menu just below, which says power level, and there is several options inside: auto, low, high, manual...

Regarding this menu, i have noticed that after every reboot the chosen option goes back to auto, and i want to fix it to low for example, can the gui save the chosen option to restore it after the reboot?

marazmista, 2020-05-11:
I'm not sure but maybe for newer cards those profiles don't apply and should be disabled. Maybe all control is now done by frequency tables, states etc.

> Regarding this menu, i have noticed that after every reboot the chosen option goes back to auto, and i want to fix it to low for example, can the gui save the chosen option to restore it after the reboot?

I will look into that.

> Also, more control with those options would be nice, to edit and create them with specific values, i cannot find any way to edit them

What do you mean by edit and create with specific values? Values available in dropdown menu are defined in kernel and user cannot add/remove/edit them:
* https://github.com/torvalds/linux/blob/ab6f762f0f53162d41497708b33c9a3236d3609e/drivers/gpu/drm/amd/amdgpu/amdgpu_pm.c#L124 
* https://github.com/torvalds/linux/blob/ab6f762f0f53162d41497708b33c9a3236d3609e/drivers/gpu/drm/amd/amdgpu/amdgpu_pm.c#L238

Akrai, 2020-05-11:
> What do you mean by edit and create with specific values? Values available in dropdown menu are defined in kernel and user cannot add/remove/edit them:
>
>     * [...]

ooh i didn't know about that, i thought it was some kind of program defaults, thanks! But manual should let you input values, shouldn't it?

marazmista, 2020-06-04:
> But manual should let you input values, shouldn't it?

Manual lets you adjust voltage and frequency states, percent overclock etc.

capoei, 2020-06-16:
there are actualy modes for newer cards with: /sys/class/drm/card0/device/pp_power_profile_mode https://dri.freedesktop.org/docs/drm/gpu/amdgpu.html
it changes behaviour how the card switches between states.
allready requested to inclued this https://github.com/marazmista/radeon-profile/issues/200 
powersave is enough for me as a non gamer and will leave my GPU much colder

marazmista, 2020-06-19:
Is `power_dpm_state` file empty when `pp_power_profile_mode` exist?

capoei, 2020-06-19:
it has "performance" i it.
"echo battery > /sys/class/drm/card0/device/power_dpm_state" has no effect.

capoei, 2020-06-19:
this is how it works,
list the modes: "cat /sys/class/drm/card0/device/pp_power_profile_mode"
echo the number of the mode "echo [numer_of_mode] > /sys/class/drm/card0/device/pp_power_profile_mode "

marazmista, 2020-07-11:
I implemented `pp_power_profile_mode` support in a50694e2a92bcd55d136a923ac44df7da8df8707 (without Custom profile for now)

closed 2020-10-04 by marazmista