marazmista/radeon-profile #187: Radeon-profile freezing on Archlinux randomly

Razer0123, 2020-03-07:
Hello, not sure why, but randomly this program makes my Archlinux setup to freeze, it even happens when deleting overclock profiles for example, so i am not sure it is an overclocking issue

How could i provide proper information to fix this?

Oxalin, 2020-03-11:
Do you have any meaningful traces under journalctl? I run Archlinux without any issues.

Also, could you provide your GPU info?

Razer0123, 2020-03-13:
The only thing i have found is posted here, not sure if it is related

https://bbs.archlinux.org/viewtopic.php?id=253457

RX 580 Nitro Sapphire

Oxalin, 2020-03-17:
Is your problem only appearing when using radeon-profile? If it is not running, is your system crashing randomly? When set to default, radeon-profile only shows your current GPU states, so it shouldn't be the culprit. Also, AMDGPU still receives fixes from time to time about RX580 (Polaris).

After a crash, could you open a console and type de following "journalctl -b -1" please? This will give you the log from the last boot. Or you could use "journalctl --since [date and time prior the crash]" which will narrow the time period you want to investigate.

Oxalin, 2020-03-17:
Also, from your bbs.archlinux.org post, if you try to underclock or overclock your card, if the values goes above or under the stable limits, your system will crash. According to your first comment (here, at the top), you were playing with overclock / underclock / undervolt. Deleting the profile doesn't change the applied settings until you apply the new selected profile. My two cents.

Razer0123, 2020-03-17:
I can assure you I have been using the same profile for years, and same values in Windows with no crash

Also, weird thing, sometimes the system crashes just saving the profile, so I suppose there’s something else

Razer0123, 2020-03-24:
> Also, from your bbs.archlinux.org post, if you try to underclock or overclock your card, if the values goes above or under the stable limits, your system will crash. According to your first comment [...]

Tried same app on Manjaro, no issues

Did you do anything specific to configure your card on Arch?

Razer0123, 2020-03-25:
Keeps happening, any way i could provide more helpful info?

Oxalin, 2020-03-30:
And you said (from an email I received) that you had no problem under Manjaro, right? If I'm right, Manjaro and ArchLinux are not exactly using the same desktop, themes and icon sets, isn't it?

Your dumps point to an error with Qt5. I'll dig this further.

Razer0123, 2020-03-31:
Yes but apart from Manjaro or Ubuntu, it also happened when saving profiles at default clock, that is why i am sure it is no overclock issue

What is the error you are referring to?

Oxalin, 2020-03-31:
Since I'm doing this on the side, it's taking me a bit of time. Using gdb with the first dump point to the following.

Core was generated by `radeon-profile'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007fe3c7ffc03e in ?? () from /usr/lib/libQt5Gui.so.5

But I'll need the Qt5 symboles to have a better idea. Your log also points in that direction.

Oxalin, 2020-03-31:
@Razer0123 Would it be possible for you to install qt5-base-debug from here: https://github.com/qutebrowser/qutebrowser/blob/master/doc/stacktrace.asciidoc#archlinux

I don't have the time needed to build a debug package for a minor upgrade (5.14.0 VS 5.14.1), so if you could either downgrade your qt5-base to 5.14.0 (look in your cached packages) or install the qt5-base-debug from the precompiled version, that would be appreciated. Then, just send me another dump.

Oxalin, 2020-04-03:
@Razer0123 Arch just pushed Qt5.14.2. I built the debug version. So you could forget installing qt5-base-debug and simply upgrade qt5-base to 5.14.2, then create a new core dump after that. I'll be able to debug it further.

Razer0123, 2020-04-04:
> @Razer0123 Arch just pushed Qt5.14.2. I built the debug version. So you could forget installing qt5-base-debug and simply upgrade qt5-base to 5.14.2, then create a new core dump after that. I'll be able to debug it further.

Had the issue only once lately, if it happens again after the update i will report back

Razer0123, 2020-04-11:
Happened again, a little bit differently, had a problem with a Gnome extensions that opened several radeonprofile instances,  but may be still useful

https://file.io/8VR2f9

closed 2020-10-02 by Razer0123