marazmista/radeon-profile #192: Feature Request: Memory Timings Level Adjustment

GM-Script-Writer-62850, 2020-03-30:
On windows wattman has this feature, it would be very nice to see it in this software
[image]

JoneKone, 2020-05-13:
> On windows wattman has this feature, it would be very nice to see it in this software
> [image]

https://github.com/Eliovp/amdmemorytweak
you can do it with this.

GM-Script-Writer-62850, 2020-05-13:
I have tried using that, it changing timings (even setting them to the same value they are) causes the system to crash, system does not crash when not using GUI, but changes only affect the current speed (idle speed)
i have tried getting the timings it uses in windows, the driver acts strange when dealing with memory on windows for example you can crank the memory frequency and power consumption goes up, but you apply nearly any change to mem speed on linux and it crashes

JoneKone, 2020-05-15:
> I have tried using that, it changing timings (even setting them to the same value they are) causes the system to crash, system does not crash when not using GUI, but changes only affect the current [...]

Yes I have locked my RX56 memory speed to constant 800Mhz cause it's change on Linux means instability, it is not implemented properly on linux. Had the same issue on RX570 (Black screen of death) It's apparently repaired by resetting the GPU and few times I have seen AMD GPU's reset properly on Linux, how ever it doesn't work properly and is implemented wrong. Keep it on Constant Memory clocks this will probably solve your issues.