marazmista/radeon-profile #200: [feature request] option to change power profile mode

capoei, 2020-05-08:
changing power profile mode (/sys/class/drm/card0/device/pp_power_profile_mode) might not be intresting for gamers, but it sure is for all the rest

marazmista, 2020-07-11:
`pp_power_profile_mode` support has been implemented in a50694e (without Custom profile for now)

capoei, 2020-07-11:
very nice

capoei, 2020-07-12:
I just installed, but I can't find it. where is it suposed to be?

marazmista, 2020-07-12:
On the left side, instead of battery/balanced/performance buttons.

capoei, 2020-07-12:
nothing there. missing icon theme?
https://drive.google.com/file/d/1HMh-VsYpgynbaE19uSR7_gpYZ--BGsbX/view?usp=sharing

marazmista, 2020-07-13:
Could you post contents of `pp_power_profile_mode`?

capoei, 2020-07-13:
`[root@bahiahost studio]# cat /sys/class/drm/card0/device/pp_power_profile_mode 
NUM        MODE_NAME     SCLK_UP_HYST   SCLK_DOWN_HYST SCLK_ACTIVE_LEVEL     MCLK_UP_HYST   MCLK_DOWN_HYST MCLK_ACTIVE_LEVEL
  0   BOOTUP_DEFAULT:        -                -                -                -                -                -
  1   3D_FULL_SCREEN:        0              100               30                0              100               10
  2   POWER_SAVING *:       10                0               30                0              100               10
  3            VIDEO:        -                -                -               10               16               31
  4               VR:        0               11               50                0              100               10
  5          COMPUTE:        0                5               30                -                -                -
  6           CUSTOM:        -                -                -                -                -      `

contet probably changes from card to card?

capoei, 2020-07-13:
[root@bahiahost studio]# cat /sys/class/drm/card0/device/pp_power_profile_mode 
NUM        MODE_NAME     SCLK_UP_HYST   SCLK_DOWN_HYST SCLK_ACTIVE_LEVEL     MCLK_UP_HYST   MCLK_DOWN_HYST MCLK_ACTIVE_LEVEL
  0   BOOTUP_DEFAULT:        -                -                -                -                -                -
  1   3D_FULL_SCREEN:        0              100               30                0              100               10
  2   POWER_SAVING *:       10                0               30                0              100               10
  3            VIDEO:        -                -                -               10               16               31
  4               VR:        0               11               50                0              100               10
  5          COMPUTE:        0                5               30                -                -                -
  6           CUSTOM:        -                -                -                -                -

marazmista, 2020-07-15:
Thanks. Yes, I guess that content depends on card, but I'm more concerned that buttons on left panel are not created. I pushed commit with some additional debug messages, could you build debug binary and paste log?

capoei, 2020-07-15:
sorry, I have no clue how git works. I use the AUR package to install on Arch

marazmista, 2020-07-15:
Please follow instructions from readme: https://github.com/marazmista/radeon-profile#build but instead of `qmake` type `qmake CONFIG+=debug`. Binary will be in `radeon-profile/target` directory and you can run it from there.

capoei, 2020-07-15:
can you modify the PKGBUILD? https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=radeon-profile-git
In Arch you are not supposed to mess with the filesystem without pacman knowing it

marazmista, 2020-07-16:
You don't have to install it. You can run it from place where it has been compiled without modifying version installed with pacman.
In PKGBUILD, if you want to install debug build, you can add `debug` at the end of line 33. (so it will look like this: `CONFIG+=silent debug`)

capoei, 2020-07-16:
So now you want the terminal output or do I have to add a switch to the comand?

this is what terminal says:

[studio@bahiahost ~]$ radeon-profile
Creating application object
Translation not found.
Creating radeon_profile
Icon theme "elementary" not found.
Icon theme "deepin" not found.
Loading configuration
Creating ui elements
Analyzing screen  0
  Analyzing output  0
  Analyzing output  1
    Analyzing active mode
      Property is EDID, parsing it
Using QCharRef with an index pointing outside the valid range of a QString. The corresponding behavior is deprecated, and will be changed in a future version of Qt.
Using QCharRef with an index pointing outside the valid range of a QString. The corresponding behavior is deprecated, and will be changed in a future version of Qt.
Using QCharRef with an index pointing outside the valid range of a QString. The corresponding behavior is deprecated, and will be changed in a future version of Qt.
Searching PnP ID:  "SAM"
  Analyzing output  2

marazmista, 2020-07-18:
Please try with latest commit, buttons should work fine now.

capoei, 2020-07-18:
So, the buttons are there, but it wont let me switch. it always goes back to BOOTUP_DEFAULT.
deamon is runing and power level is in manual

marazmista, 2020-07-18:
I pushed a commit with a fix: https://github.com/marazmista/radeon-profile/commit/2a49ea5f5d88d7b8fb21fb6ea54c49f05166c314

capoei, 2020-07-18:
yep. that fixed it. nice

closed 2020-10-04 by marazmista