marazmista/radeon-profile #201: Radeon Profile on AppImage

alexwender1, 2020-05-10:
Can you make the application available in the AppImage format? It is easier to run on any system, as it is a containerized and executable format.

pudzianwrc, 2020-05-14:
I compiled Radeon Profile today on Ubuntu 20.04 64-bit.

https://drive.google.com/open?id=1EPZXcdF6mPFesrXdl32lmYYPbmEMJMAw

alexwender1, 2020-05-15:
> I compiled and created AppImage Radeon Profile today on Ubuntu 20.04 64-bit.
> 
> https://drive.google.com/open?id=1EPZXcdF6mPFesrXdl32lmYYPbmEMJMAw

Thanks for responding and trying to help me. Unfortunately, a problem has occurred.

I'm using deepin 20 BETA (based on debian Buster), which has the _libm.so.6_ package in version **2.28.7**; but your AppImage requires version **2.29** (corresponding to Ubuntu LTS). I tried to update manually; but incompatibilities with other system complements occur.

The best option is to have AppImage based on debian stable; which will be compatible with most derived distributions (including Ubuntu itself) on debian. @marazmista 

[image]

pudzianwrc, 2020-05-15:
I compiled everything under Debian buster VM. You have radeon-profile-daemon precompiled (uncompress, enter to the catalog and execute "sudo make install") ass wel. Try it now. 

https://drive.google.com/open?id=1xUxax63-VhIFbPQU-g8ELB8G63CKpI2A

alexwender1, 2020-05-16:
> I compiled everything under Debian buster VM. You have radeon-profile-daemon precompiled (uncompress, enter to the catalog and execute "sudo make install") ass wel. Try it now.
> 
> https://drive.google.com/open?id=1xUxax63-VhIFbPQU-g8ELB8G63CKpI2A

I am very grateful for the concern. During the "sudo make install" there were 2 problems (Error 127 and then Error 3); I solved it by installing the make and qt5-default packages, respectively.

I believe that the installation occurred correctly:

[image]

The application is currently displayed in this way (image below). I believe that my GPU may be incompatible; I don't know if there is a list of compatible graphics cards.

[image]

I reaffirm my thanks for your attention and especially for your patience in trying to help.

pudzianwrc, 2020-05-16:
What exactly is your graphics card model? Ps. After installation
radeon-profile-daemon execute: 
sudo systemctl enable radeon-profile-daemon.service 
and 
sudo systemctl start radeon-profile-daemon.service 

to make the daemon running.

https://github.com/marazmista/radeon-profile-daemon

alexwender1, 2020-05-16:
Something has changed here ;p. I think it's working.

[image]