marazmista/radeon-profile #202: radeon-profile-daemon.service has a bad unit file setting

bassamanator, 2020-05-12:
The app runs fine but there is no information that is displayed. I think it's because the daemon has is not starting.
```
$ systemctl status radeon-profile-daemon
● radeon-profile-daemon.service
     Loaded: bad-setting (Reason: Unit radeon-profile-daemon.service has a bad unit file setting.)
     Active: inactive (dead)

May 11 21:39:08 bassam-pc systemd[1]: /etc/systemd/system/radeon-profile-daemon.service:1308: Assignment outside of section. Ignoring.
May 11 21:39:08 bassam-pc systemd[1]: /etc/systemd/system/radeon-profile-daemon.service:1309: Assignment outside of section. Ignoring.
May 11 21:39:08 bassam-pc systemd[1]: /etc/systemd/system/radeon-profile-daemon.service:1310: Assignment outside of section. Ignoring.
May 11 21:39:08 bassam-pc systemd[1]: /etc/systemd/system/radeon-profile-daemon.service:1311: Assignment outside of section. Ignoring.
May 11 21:39:08 bassam-pc systemd[1]: /etc/systemd/system/radeon-profile-daemon.service:1313: Assignment outside of section. Ignoring.
May 11 21:39:08 bassam-pc systemd[1]: /etc/systemd/system/radeon-profile-daemon.service:1314: Assignment outside of section. Ignoring.
May 11 21:39:08 bassam-pc systemd[1]: /etc/systemd/system/radeon-profile-daemon.service:1315: Assignment outside of section. Ignoring.
May 11 21:39:08 bassam-pc systemd[1]: /etc/systemd/system/radeon-profile-daemon.service:1316: Assignment outside of section. Ignoring.
May 11 21:39:08 bassam-pc systemd[1]: /etc/systemd/system/radeon-profile-daemon.service:1319: Assignment outside of section. Ignoring.
May 11 21:39:08 bassam-pc systemd[1]: /etc/systemd/system/radeon-profile-daemon.service:1320: Assignment outside of section. Ignoring.

```

marazmista, 2020-05-13:
How did you install daemon? Can you provide `radeon-profile-daemon.service` file?

bassamanator, 2020-05-14:
@marazmista I installed it manually. Copied it to the systemd/system if I remember correctly (it was in the instructions on the git page.)

Should I delete it? And install it via...?

marazmista, 2020-05-16:
Have you modified file somehow? I've only found similar  issue and is related to wrong encoding: https://stackoverflow.com/questions/42286099/systemd-script-assignment-outside-of-section-missing/42288033#42288033

In Arch linux it is copied to `/usr/lib/systemd/system/`

bassamanator, 2020-05-18:
@marazmista So i deleted the daemon and installed via the instructions instead of manually copying it over.

It's working now! Great app, been looking for this on linux for along time now. Thanks!

closed 2020-06-04 by marazmista