marazmista/radeon-profile #206: New segfaults on debian/sid

Cyclic3, 2020-05-18:
After a year of flawless performance, radeon-profile has suddenly started segfaulting whenever I try to start it again.

GDB backtrace:
<pre>#0  <font color="#3465A4">0x00007ffff7bd1848</font> in <font color="#C4A000">QtCharts::AbstractChartLayout::sizeHint(Qt::SizeHint, QSizeF const&amp;) const</font> () at <font color="#4E9A06">/lib/x86_64-linux-gnu/libQt5Charts.so.5</font>
#1  <font color="#3465A4">0x00007ffff787cd71</font> in  () at <font color="#4E9A06">/lib/x86_64-linux-gnu/libQt5Widgets.so.5</font>
#2  <font color="#3465A4">0x00007ffff787d746</font> in <font color="#C4A000">QGraphicsLayoutItem::effectiveSizeHint(Qt::SizeHint, QSizeF const&amp;) const</font> () at <font color="#4E9A06">/lib/x86_64-linux-gnu/libQt5Widgets.so.5</font>
#3  <font color="#3465A4">0x00007ffff78bf199</font> in <font color="#C4A000">QGraphicsWidget::sizeHint(Qt::SizeHint, QSizeF const&amp;) const</font> () at <font color="#4E9A06">/lib/x86_64-linux-gnu/libQt5Widgets.so.5</font>
#4  <font color="#3465A4">0x00007ffff787cf21</font> in  () at <font color="#4E9A06">/lib/x86_64-linux-gnu/libQt5Widgets.so.5</font>
#5  <font color="#3465A4">0x00007ffff787d746</font> in <font color="#C4A000">QGraphicsLayoutItem::effectiveSizeHint(Qt::SizeHint, QSizeF const&amp;) const</font> () at <font color="#4E9A06">/lib/x86_64-linux-gnu/libQt5Widgets.so.5</font>
#6  <font color="#3465A4">0x00007ffff78c20c6</font> in <font color="#C4A000">QGraphicsWidget::setGeometry(QRectF const&amp;)</font> () at <font color="#4E9A06">/lib/x86_64-linux-gnu/libQt5Widgets.so.5</font>
#7  <font color="#3465A4">0x00007ffff78bf728</font> in <font color="#C4A000">QGraphicsWidget::resize(QSizeF const&amp;)</font> () at <font color="#4E9A06">/lib/x86_64-linux-gnu/libQt5Widgets.so.5</font>
#8  <font color="#3465A4">0x00007ffff78c2dce</font> in <font color="#C4A000">QGraphicsWidget::updateGeometry()</font> () at <font color="#4E9A06">/lib/x86_64-linux-gnu/libQt5Widgets.so.5</font>
#9  <font color="#3465A4">0x000055555558f525</font> in <font color="#C4A000">PieProgressBar::init()</font> ()
#10 <font color="#3465A4">0x000055555559bebf</font> in <font color="#C4A000">TopbarManager::addToLayout(QHBoxLayout*, TopbarItemDefinitionSchema const&amp;)</font> ()
#11 <font color="#3465A4">0x000055555557d0c2</font> in <font color="#C4A000">radeon_profile::setupDeviceDependantUiElements()</font> ()
#12 <font color="#3465A4">0x000055555557d292</font> in <font color="#C4A000">radeon_profile::initializeDevice()</font> ()
#13 <font color="#3465A4">0x000055555557d8c1</font> in <font color="#C4A000">radeon_profile::daemonConnected()</font> ()
#14 <font color="#3465A4">0x00005555555ef1ce</font> in <font color="#C4A000">radeon_profile::qt_static_metacall(QObject*, QMetaObject::Call, int, void**)</font> ()
#15 <font color="#3465A4">0x00007ffff69e0388</font> in <font color="#C4A000">QMetaObject::activate(QObject*, int, int, void**)</font> () at <font color="#4E9A06">/lib/x86_64-linux-gnu/libQt5Core.so.5</font>
#16 <font color="#3465A4">0x00007ffff6d73738</font> in  () at <font color="#4E9A06">/lib/x86_64-linux-gnu/libQt5Network.so.5</font>
#17 <font color="#3465A4">0x00007ffff6d73c79</font> in <font color="#C4A000">QLocalSocket::connectToServer(QFlags&lt;QIODevice::OpenModeFlag&gt;)</font> () at <font color="#4E9A06">/lib/x86_64-linux-gnu/libQt5Network.so.5</font>
#18 <font color="#3465A4">0x00005555555c5353</font> in <font color="#C4A000">DaemonComm::connectToDaemon()</font> ()
#19 <font color="#3465A4">0x000055555557d7e1</font> in <font color="#C4A000">radeon_profile::radeon_profile(QWidget*)</font> ()
#20 <font color="#3465A4">0x0000555555577035</font> in <font color="#C4A000">main</font> ()</pre>

Any ideas?

Cyclic3, 2020-05-18:
Just to point out that it starts fine when run as non-root _without_ the daemon running (just doesn't show any cards), but I get a similar segfault when I run it as root without the daemon:
```
Creating application object
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
Creating radeon_profile
Loading configuration
Creating ui elements
Analyzing screen  0
  Analyzing output  0
    Analyzing active mode
      Property is EDID, parsing it
Searching PnP ID:  "AUS"
  Analyzing output  1
  Analyzing output  2
  Analyzing output  3
Initializing device
Card detected (amdgpu):  "card0"
hwmon path:  "/sys/class/drm/card0/device/hwmon/hwmon0/"
Opened /dev/dri/renderD128 , fd number: 16
ioctlHandler: everything ok
Power method: DPM
ioctlHandler: everything ok
GPU max core clk:  1590 
 max mem clk:  800 
 vram size:  8059.62
Segmentation fault
```
#0  0x00007ffff7bd1848 in QtCharts::AbstractChartLayout::sizeHint(Qt::SizeHint, QSizeF const&) const () from /lib/x86_64-linux-gnu/libQt5Charts.so.5
#1  0x00007ffff787cd71 in ?? () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#2  0x00007ffff787d746 in QGraphicsLayoutItem::effectiveSizeHint(Qt::SizeHint, QSizeF const&) const () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#3  0x00007ffff78bf199 in QGraphicsWidget::sizeHint(Qt::SizeHint, QSizeF const&) const () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#4  0x00007ffff787cf21 in ?? () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#5  0x00007ffff787d746 in QGraphicsLayoutItem::effectiveSizeHint(Qt::SizeHint, QSizeF const&) const () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#6  0x00007ffff78c20c6 in QGraphicsWidget::setGeometry(QRectF const&) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#7  0x00007ffff78bf728 in QGraphicsWidget::resize(QSizeF const&) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#8  0x00007ffff78c2dce in QGraphicsWidget::updateGeometry() () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#9  0x0000555555578801 in QGraphicsLayoutItem::setMinimumSize (this=0x555555d96b50, aw=0, ah=0) at /usr/include/x86_64-linux-gnu/qt5/QtWidgets/qgraphicslayoutitem.h:124
#10 0x000055555557a3db in PieProgressBar::init (this=0x555555d96ae0) at components/pieprogressbar.h:118
#11 0x0000555555579614 in PieProgressBar::PieProgressBar (this=0x555555d96ae0, max=100, id=FAN_SPEED_PERCENT, fillColor=..., parent=0x0) at components/pieprogressbar.h:36
#12 0x000055555557b263 in PieItem::PieItem (this=0x555555d96ae0, max=100, vId=FAN_SPEED_PERCENT, fillColor=..., parent=0x0) at components/topbarcomponents.h:127
#13 0x000055555557b914 in TopbarManager::addToLayout (this=0x7fffffffe278, layout=0x5555556f1690, tis=...) at components/topbarcomponents.h:240
#14 0x000055555557b745 in TopbarManager::createTopbar (this=0x7fffffffe278, layout=0x5555556f1690) at components/topbarcomponents.h:216
#15 0x00005555555720ea in radeon_profile::setupDeviceDependantUiElements (this=0x7fffffffe150) at radeon_profile.cpp:163
#16 0x00005555555716cb in radeon_profile::initializeDevice (this=0x7fffffffe150) at radeon_profile.cpp:85
#17 0x00005555555711bf in radeon_profile::radeon_profile (this=0x7fffffffe150, parent=0x0) at radeon_profile.cpp:58
#18 0x00005555555703dd in main (argc=1, argv=0x7fffffffe508) at main.cpp:24
```

Cyclic3, 2020-05-18:
Ok, I don't know why, but a combination of pulling out and replacing my hardware, rebooting, and rebuilding multiple times made it work.

closed 2020-05-18 by Cyclic3

marazmista, 2020-05-19:
Strange.
If this happen again, you can try start it with default configuration. To do this, remove or change name of directory: `~/.config/radeon-profile`