marazmista/radeon-profile #209: Fan speed graph often does not work

djrscally, 2020-06-01:
Hello

Cool tool; thanks for making it. 

I quite often find that the fan speed gauge doesn't work; it just stays at 0% all the time. The actual fan control still works fine; it'll follow my custom curves or set to a fixed value I give it, it's just the gauge that doesn't work. 

[image]

Unfortunately I can't quite spot where the program is logging to so I can't find any diagnostics. Launching from terminal doesn't give me any errors about the graphs. If I close and reopen the program a bunch of times, eventually the speed control graph will decide it's happy and work.

I'm aware this is a bit barebones; if you can give me a bit of guidance on finding some logging info I'll dig into it some more.

marazmista, 2020-06-04:
Does it show fan speed in upper list, below GPU Vram usage, on GPU Summary tab?

djrscally, 2020-06-04:
There's a row for Fan Speed as a percentage and Fan Speed RPM, but both are set to 0 regardless of the actual setting.

Akrai, 2020-06-05:
I have noticed that, for example after a reboot, even when the program is set to keep using the same fan curve, and when i open the gui it says it is using it, it is in fact not using it, and i can fix it by going to fan control, and changing from custom curve to any of fixed or auto, and then coming back to custom, and since then it uses the custom curve until the next reboot

djrscally, 2020-06-05:
Probably unrelated; I'm on a fresh OS install and haven't set up my custom
curves yet, and even if I go into fan control and set 100% the display will
still show 0%

marazmista, 2020-06-08:
Strange. I would expect no fan control whatsoever if something is not initialized correctly. Can you try build debug binary and start from console? Instead of `qmake` you need to type `qmake CONFIG+=debug`

When you start radeon-profile, it should print messages like:
```
Initializing device
Card detected (radeon):  "card0"
hwmon path:  "/sys/class/drm/card0/device/hwmon/hwmon2/"
...
Fan control available
```

djrscally, 2020-06-08:
Yep:

>Creating application object
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
Creating radeon_profile
Loading configuration
Creating ui elements
Analyzing screen  0
  Analyzing output  0
    Analyzing active mode
      Property is EDID, parsing it
Searching PnP ID:  "ACI"
  Analyzing output  1
  Analyzing output  2
  Analyzing output  3
    Analyzing active mode
      Property is EDID, parsing it
Searching PnP ID:  "ACI"
Initializing device
Card detected (amdgpu):  "card0"
hwmon path:  "/sys/class/drm/card0/device/hwmon/hwmon0/"
Opened /dev/dri/renderD128 , fd number: 16
ioctlHandler: everything ok
Power method: DPM
ioctlHandler: everything ok
GPU max core clk:  1780 
 max mem clk:  875 
 vram size:  6091.37
Handling found device features
Fan control available

I tried setting fixed speed 100%; the card ramped up to MAXIMUM POWER as always, but no further messages were printed to console.

djrscally, 2020-06-09:
Attempting to trace through the code and slightly confused about what's happening. It looks to me (I'm not familiar with cpp but I'm pretty sure this is right) that both reading the setting and changing the setting involves interacting with /sys/class/drm/card0/device/hwmon/hwmon0/pwm1 - is that right? When I cat that file, it shows 0, even when the setting is 100%. That doesn't really make sense; radeon-profile **is** controlling the cards fan speed, so if all the adjustGpuFan() function does is write the new speed to that file then I don't see how it's working.

marazmista, 2020-06-09:
> It looks to me (I'm not familiar with cpp but I'm pretty sure this is right) that both reading the setting and changing the setting involves interacting with /sys/class/drm/card0/device/hwmon/hwmon0/pwm1 - is that right? 

Correct. Reading and setting fan speed is controlled by this file. Value range is 0-255.
If there is always 0 then, to me, it looks like there may be a bug in driver because file should report something different than 0, when fan is spinning.

djrscally, 2020-06-09:
Yep. So this occassion I boot, the fan speed graph is happy aaaaand:

```
djrscally@valhalla:~$ cat /sys/class/drm/card0/device/hwmon/hwmon0/pwm1
53
```

There's a bug here, but I don't think it's yours.

bamyasi, 2020-08-05:
Same here: Radeon RX 5500XT 8GB, fan RPM permanently at zero both in radeon-profile and when monitored via sensors command. Even if I force it to fixed on the fan profile tab, the fans do actually start to rotate but still reading is zero RPM, by sensors command and by the program.

Looks like a driver kernel/bug to me.

$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.1 LTS
Release:        20.04
Codename:       focal

$ uname -a
Linux xxxxxx 5.4.0-42-generic #46-Ubuntu SMP Fri Jul 10 00:24:02 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

djrscally, 2020-08-05:
Agree, it's definitely a kernel bug somewhere. I reported it [here] - probably worth a "yeah, me too" there as well. I traced it back in the driver code as much as my shaky C knowledge would allow, but concluded it was possibly firmware related and so immediately gave in.

djrscally, 2020-11-20:
Just an update on this; there was a patch submitted that seemed to fix the issue on drm/amdgpu. It's been reverted because there were some regressions, but I expect it'll be permanently resolved soon.