marazmista/radeon-profile #212: Failing to build on current Manjaro

gardotd426, 2020-06-15:
I've been using radeon-profile for almost a year, but on a brand new Manjaro install, it fails to build:
```
./globalStuff.h:397:41: warning: ‘void QProcess::start(const QString&, QIODevice::OpenMode)’ is deprecated: Use QProcess::start(const QString &program, const QStringList &arguments,OpenMode mode = ReadWrite) instead [-Wdeprecated-declarations]
  397 |         p->start(cmd,QIODevice::ReadOnly);
      |                                         ^
In file included from /usr/include/qt/QtCore/QProcess:1,
                 from ./globalStuff.h:8,
                 from ./components/topbarcomponents.h:7,
                 from dialogs/dialog_topbarcfg.h:9,
                 from dialogs/dialog_topbarcfg.cpp:4:
/usr/include/qt/QtCore/qprocess.h:168:10: note: declared here
  168 |     void start(const QString &command, OpenMode mode = ReadWrite);
      |          ^~~~~
execbin.cpp: In member function ‘void ExecBin::runBin(const QString&)’:
execbin.cpp:62:17: warning: ‘void QProcess::start(const QString&, QIODevice::OpenMode)’ is deprecated: Use QProcess::start(const QString &program, const QStringList &arguments,OpenMode mode = ReadWrite) instead [-Wdeprecated-declarations]
   62 |     p->start(cmd);
      |                 ^
In file included from /usr/include/qt/QtCore/QProcess:1,
                 from execbin.h:10,
                 from execbin.cpp:4:
/usr/include/qt/QtCore/qprocess.h:168:10: note: declared here
  168 |     void start(const QString &command, OpenMode mode = ReadWrite);
      |          ^~~~~
In file included from dialogs/dialog_defineplot.h:5,
                 from dialogs/dialog_defineplot.cpp:1:
./globalStuff.h: In static member function ‘static QStringList globalStuff::grabSystemInfo(QString)’:
./globalStuff.h:384:41: warning: ‘void QProcess::start(const QString&, QIODevice::OpenMode)’ is deprecated: Use QProcess::start(const QString &program, const QStringList &arguments,OpenMode mode = ReadWrite) instead [-Wdeprecated-declarations]
  384 |         p->start(cmd,QIODevice::ReadOnly);
      |                                         ^
In file included from /usr/include/qt/QtCore/QProcess:1,
                 from ./globalStuff.h:8,
                 from dialogs/dialog_defineplot.h:5,
                 from dialogs/dialog_defineplot.cpp:1:
/usr/include/qt/QtCore/qprocess.h:168:10: note: declared here
  168 |     void start(const QString &command, OpenMode mode = ReadWrite);
      |          ^~~~~
In file included from dialogs/dialog_defineplot.h:5,
                 from dialogs/dialog_defineplot.cpp:1:
./globalStuff.h: In static member function ‘static QStringList globalStuff::grabSystemInfo(QString, QProcessEnvironment)’:
./globalStuff.h:397:41: warning: ‘void QProcess::start(const QString&, QIODevice::OpenMode)’ is deprecated: Use QProcess::start(const QString &program, const QStringList &arguments,OpenMode mode = ReadWrite) instead [-Wdeprecated-declarations]
  397 |         p->start(cmd,QIODevice::ReadOnly);
      |                                         ^
```
This is just a segment of the hundreds of errors, it actually exceeded my terminal scrollback even. 

Here's the last few errors:
```
dialogs/dialog_defineplot.cpp:135:66: warning: ‘void QTreeWidgetItem::setBackgroundColor(int, const QColor&)’ is deprecated: Use QTreeWidgetItem::setBackground() instead [-Wdeprecated-declarations]
  135 |     item->setBackgroundColor(1,getColor(item->backgroundColor(1)));
      |                                                                  ^
In file included from /usr/include/qt/QtWidgets/QtWidgets:297,
                 from /usr/include/qt/QtCharts/QtChartsDepends:5,
                 from /usr/include/qt/QtCharts/QtCharts:3,
                 from ./components/rpplot.h:8,
                 from dialogs/dialog_defineplot.h:6,
                 from dialogs/dialog_defineplot.cpp:1:
/usr/include/qt/QtWidgets/qtreewidget.h:142:17: note: declared here
  142 |     inline void setBackgroundColor(int column, const QColor &color)
      |                 ^~~~~~~~~~~~~~~~~~
dialogs/dialog_defineplot.cpp: In member function ‘void Dialog_definePlot::addSelectedItemToSchema(int, QTreeWidgetItem*, QMap<ValueID, QColor>&)’:
dialogs/dialog_defineplot.cpp:165:98: warning: ‘QColor QTreeWidgetItem::backgroundColor(int) const’ is deprecated: Use QTreeWidgetItem::background() instead [-Wdeprecated-declarations]
  165 |             schemaDataList.insert(listRelationToValueID.value(itemIndex), item->backgroundColor(1));
      |                                                                                                  ^
In file included from /usr/include/qt/QtWidgets/QtWidgets:297,
                 from /usr/include/qt/QtCharts/QtChartsDepends:5,
                 from /usr/include/qt/QtCharts/QtCharts:3,
                 from ./components/rpplot.h:8,
                 from dialogs/dialog_defineplot.h:6,
                 from dialogs/dialog_defineplot.cpp:1:
/usr/include/qt/QtWidgets/qtreewidget.h:139:19: note: declared here
  139 |     inline QColor backgroundColor(int column) const
      |                   ^~~~~~~~~~~~~~~
dialogs/dialog_defineplot.cpp: In member function ‘void Dialog_definePlot::on_btn_save_clicked()’:
dialogs/dialog_defineplot.cpp:194:68: warning: ‘const QBrush& QPalette::background() const’ is deprecated: Use QPalette::window() instead [-Wdeprecated-declarations]
  194 |     schema.background = ui->frame_background->palette().background().color();
      |                                                                    ^
In file included from /usr/include/qt/QtWidgets/qwidget.h:48,
                 from /usr/include/qt/QtWidgets/qdialog.h:44,
                 from /usr/include/qt/QtWidgets/QDialog:1,
                 from dialogs/dialog_defineplot.h:4,
                 from dialogs/dialog_defineplot.cpp:1:
/usr/include/qt/QtGui/qpalette.h:149:26: note: declared here
  149 |     inline const QBrush &background() const { return window(); }
      |                          ^~~~~~~~~~
dialogs/dialog_defineplot.cpp:212:78: warning: ‘const QBrush& QPalette::background() const’ is deprecated: Use QPalette::window() instead [-Wdeprecated-declarations]
  212 |     schema.left.penGrid = QPen(ui->frame_leftGridColor->palette().background().color(),
      |                                                                              ^
In file included from /usr/include/qt/QtWidgets/qwidget.h:48,
                 from /usr/include/qt/QtWidgets/qdialog.h:44,
                 from /usr/include/qt/QtWidgets/QDialog:1,
                 from dialogs/dialog_defineplot.h:4,
                 from dialogs/dialog_defineplot.cpp:1:
/usr/include/qt/QtGui/qpalette.h:149:26: note: declared here
  149 |     inline const QBrush &background() const { return window(); }
      |                          ^~~~~~~~~~
dialogs/dialog_defineplot.cpp:216:80: warning: ‘const QBrush& QPalette::background() const’ is deprecated: Use QPalette::window() instead [-Wdeprecated-declarations]
  216 |     schema.right.penGrid = QPen(ui->frame_rightGridColor->palette().background().color(),
      |                                                                                ^
In file included from /usr/include/qt/QtWidgets/qwidget.h:48,
                 from /usr/include/qt/QtWidgets/qdialog.h:44,
                 from /usr/include/qt/QtWidgets/QDialog:1,
                 from dialogs/dialog_defineplot.h:4,
                 from dialogs/dialog_defineplot.cpp:1:
/usr/include/qt/QtGui/qpalette.h:149:26: note: declared here
  149 |     inline const QBrush &background() const { return window(); }
      |                          ^~~~~~~~~~
==> ERROR: A failure occurred in build().
    Aborting...
```
I've tried with both the newest GCC and with downgrading to 9.3.0 (some other C programs got borked after the move to GCC 10). No luck.

marazmista, 2020-06-16:
In Qt 5.15  they apparently deprecated some methods, hence the warnings. But I don't see compilation error here, Any chance to get full log?

gardotd426, 2020-06-16:
Something is wrong with the `PKGBUILD` that's causing it to treat those warnings as actual errors, because at the end it does indeed fail with `Build failed. Aborting....`. I tried it about 6 times. But I also have this repo itself cloned in my `~/gits` directory, so I went in there, did a `git pull`, and tried to build it from source myself, and everything worked (though I got what looked like the same errors). 

But the thing is, with the AUR/PKGBUILD, it was hard failing because it wouldn't even build the whole package, it would fail before the package actually got built. I've almost always used the AUR in the past for installation and this was never an issue, it must just be something broken in the PKGBUILD. I'll go ahead and close this since it's not `radeon-profile`'s fault.

closed 2020-06-16 by gardotd426