marazmista/radeon-profile #213: QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root' Translation not found.

Magiceye04, 2020-06-18:
Hello,
after installing the program via repository ppa:trebelnik-stefina/radeon-profile it does not start.
instead i get this message:

sudo -H radeon-profile  or just  sudo radeon-profile
**QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
Translation not found.**

how can I fix this problem?

Starting via menu results also in "nothing".

PC: Ryzen 1700 + Radeon VII + Ubuntu 18LTS

Best Regards
Magiceye

marazmista, 2020-06-20:
First one: seems like you don't have set `XDG_RUNTIME_DIR` environment variable hence the message. I don't think that this cause any problems.
Second: app is not translated to your locale and it defaults to English.

Try build as debug and start. Maybe there will be more additional info why it won't start.

muzena, 2020-09-04:
@Magiceye04 
Read README.md file, Installation->Ubuntu
Or visit this link [Radeon Profile team](https://launchpad.net/~radeon-profile)