marazmista/radeon-profile #216: Doesn't change the frequencies if fullscreen with non native resolution

Damjan94, 2020-06-22:
I compiled it from source on Debian, running it with sudo.
when I select overclock -> manual frequency control and select the frequency that I want
[image]

if the game(X4 foundations) is in full screen, but with the resolution set to something other than the default monitor resolution, the frequency doesn't apply in-game

The black line on the graph is the frequency. Notice how it goes all the way to the top at first, then I alt + tab(the frequency hangs at ~300Mhz ) and when I set the game to windowed mode and re-apply the frequencies, it caps at a lower frequency(632 Mhz).

[image]

While I haven't tried this with any other game, I do believe this is not game specific...

marazmista, 2020-06-28:
radoen-profile doesn't adjust frequencies and states on its own. It is basically just echo with selected states into `pp_dpm_sclk`. Could be a bug in driver, how it handles full screen applications or something.
You can try setting states manually, without using radeon-profile, this way: `echo "0 1" >  /sys/class/drm/card0/device/pp_dpm_sclk`and check with other tool how frequencies change.

evolze, 2020-07-22:
@Damjan94 I do apologize for straying off topic, but I was wondering how you were able to build radeon-profile on Debian. I followed the instructions and qmake & make both worked. Everything worked without any errors and I even compiled and built the daemon as well. 

Is there something that I'm doing wrong? Could you please provide some guidance on how you were able to build it on Debian? I'm running Debian 10 Buster (stable) with the non free repos (if that helps).

Damjan94, 2020-11-12:
@evolze sorry for the late reply, if you compiled it, and it gave no errors, the executable is in the "target" directory.
so, from the build directory I just run ```sudo ./target/radeon-profile``` to start the client without the daemon. if you have the deamon running, I suppose you just run it without sudo.
But I prefer to just run it as root every time my pc boots up(too lazy to properly set up the deamon xD )

I literally just typed the 4 commands that are stated in the [Build](https://github.com/marazmista/radeon-profile#build) section.
Obviously, you need ```build-essential```, and whatever other libraries are needed, but if you didn't have those, you would be getting errors...