marazmista/radeon-profile #217: AMD 5700 xt very low memory frequencies

EndrII, 2020-07-08:
Hi 
I noticed one very unpleasant feature and I can’t understand what the problem may be,
The manufacturer of the video card claims that the memory frequency should be at least 3500 MHz, but in fact I get only 875 Mhz
Tell me what could be the problem ?

my OS ubuntu 20.04 
Driver info :
[image]

[image]

Although in windows on the same machine, the memory frequency seems to be higher than 2000 mhz.

is it a data display problem for your utility or a driver problem?

I tried to check the frequencies with drivers from the official AMD website and with free mesa drivers. In both cases, I get a very low memory frequency (under load).

marazmista, 2020-07-15:
Values in this tables comes from sysfs files (`pp_dpm_sclk` and `pp_dpm_mclk`) so if they don't match with the specs I would think that there is something wrong with the driver.

EndrII, 2020-07-16:
i am duplicate this issue in [mesa forum ](https://gitlab.freedesktop.org/mesa/mesa/-/issues/3280). Join the discussion )

EndrII, 2020-07-17:
People from mesa explained how to correctly count the effective frequency in [this issue](https://gitlab.freedesktop.org/drm/amd/-/issues/1227)

```
We expose the actual memory controller clock rate in Linux, not the effective memory clock of the DRAMs. To translate it, it follows the following formula:

Clock conversion (Mhz):

HBM: effective_memory_clock = memory_controller_clock \* 1

G5: effective_memory_clock = memory_controller_clock \* 1

G6: effective_memory_clock = memory_controller_clock \* 2

DRAM data rate (MT/s):

HBM: effective_memory_clock \* 2 = data_rate

G5: effective_memory_clock \* 4 = data_rate

G6: effective_memory_clock \* 8 = data_rate

Bandwidth (MB/s):

data_rate \* vram_bit_width / 8 = memory_bandwidth

Some examples:

G5 on RX460:

memory_controller_clock = 1750 Mhz

effective_memory_clock = 1750 Mhz \* 1 = 1750 Mhz

data rate = 1750 \* 4 = 7000 MT/s

memory_bandwidth = 7000 \* 128 bits / 8 = 112000 MB/s

G6 on RX5700:

memory_controller_clock = 875 Mhz

effective_memory_clock = 875 Mhz \* 2 = 1750 Mhz

data rate = 1750 \* 8 = 14000 MT/s

memory_bandwidth = 14000 \* 256 bits / 8 = 448000 MB/s
```

some-username-here1, 2020-07-23:
Has this been patched in yet, by any chance? I don't really know a lot about coding, so I'm unable to help out

EndrII, 2020-07-30:
@some-username-here1 this is mistake on the view components of the radeon-profile. the driver of 5700 XT work on linux correctly.

Shijikori, 2021-01-14:
doesn't the driver report the effective clock speed rather than the multiplied clock speed? Like you have 2400MHz DDR4 which's effective speed is 1200MHz.

closed 2021-02-01 by EndrII