marazmista/radeon-profile #218: Segmentation Fault when running as root

largewedge, 2020-07-11:
Hi,

I'm running Debian Sid, kernel version 5.7.0-1 on an AMD Ryzen 3600 with an RX 5600 XT. Running qmake and make is successful with no errors but with some warnings during make. I am able to run radeon-profile as a non-root user, but it can't see my GPU, but running it as root, I get a segfault:

```
sudo ./radeon-profile 
WARNING: CPU random generator seem to be failing, disable hardware random number generation
WARNING: RDRND generated: 0xffffffff 0xffffffff 0xffffffff 0xffffffff
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
QFSFileEngine::open: No file name specified
Segmentation fault
```
 
The lines about the CPU random number generator also appear when running as non-root user (and similar warnings in grub when I boot); I believe this is just an issue with my CPU and linux generally. 

Here's an output from gdb backtrace:

```
Thread 1 "radeon-profile" received signal SIGSEGV, Segmentation fault.
0x00007ffff7663fbd in QAbstractButton::setChecked(bool) ()
   from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
(gdb) backtrace 
#0  0x00007ffff7663fbd in QAbstractButton::setChecked(bool) ()
   from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#1  0x000055555557a72c in radeon_profile::refreshUI() ()
#2  0x000055555557d521 in radeon_profile::initializeDevice() ()
#3  0x000055555557da26 in radeon_profile::radeon_profile(QWidget*) ()
#4  0x00005555555771e5 in main ()

```
As far as I can tell, all of the dependencies as listed for Debian/Ubuntu are installed as specified in the readme. Thanks!

marazmista, 2020-07-11:
Please try build and run after commit 5234901d474a8481593f63d348969b5aac551fe9

largewedge, 2020-07-11:
> Please try build and run after commit [5234901](https://github.com/marazmista/radeon-profile/commit/5234901d474a8481593f63d348969b5aac551fe9)

It looks like that fixed it. It still outputs the other lines about QStandardPaths and QFSFileEngine, but it doesn't segfault anymore.

marazmista, 2020-07-15:
I edited the sizepolicy for that dropdown menu.
Regarding the clock speeds buttons on left, I pushed commit with some additional debug messages, could you build debug binary and paste log?

largewedge, 2020-07-16:
Here's the output just from running sudo ./radeon-profile:

```
WARNING: CPU random generator seem to be failing, disable hardware random number generation
WARNING: RDRND generated: 0xffffffff 0xffffffff 0xffffffff 0xffffffff
Creating application object
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
Creating radeon_profile
Loading configuration
Creating ui elements
Analyzing screen  0
  Analyzing output  0
  Analyzing output  1
  Analyzing output  2
  Analyzing output  3
    Analyzing active mode
      Property is EDID, parsing it
Using QCharRef with an index pointing outside the valid range of a QString. The corresponding behavior is deprecated, and will be changed in a future version of Qt.
Using QCharRef with an index pointing outside the valid range of a QString. The corresponding behavior is deprecated, and will be changed in a future version of Qt.
Using QCharRef with an index pointing outside the valid range of a QString. The corresponding behavior is deprecated, and will be changed in a future version of Qt.
Searching PnP ID:  "AOC"
Initializing device
Card detected:
 module:  "amdgpu" 
 sysName(path):  "card0" 
 name:  "Advanced Micro Devices, Inc. [AMD/ATI] Navi 10 [Radeon RX 5600 OEM/5600 XT / 5700/5700 XT] (rev ca)"
hwmon path:  "/sys/class/drm/card0/device/hwmon/hwmon0/"
Opened /dev/dri/renderD128 , fd number: 15
QFSFileEngine::open: No file name specified
Detected power method based on sysfs power_method file: 3
Power method: Power Profile Modes. Creating profiles list
ioctlHandler: everything ok
ioctlHandler: everything ok
GPU max core clk:  1780 
 max mem clk:  875 
 vram size:  5991.77
Handling found device features
Creating power profiles control buttons
Fan control available

```

largewedge, 2020-07-16:
This is what it looks like, no buttons:

[image]

marazmista, 2020-07-16:
I pushed a commit (https://github.com/marazmista/radeon-profile/commit/8c637c3b1272be18d8f56c5ac358b28986607e2b) that I think fixes this and buttons should show up now.

largewedge, 2020-07-16:
> I pushed a commit ([8c637c3](https://github.com/marazmista/radeon-profile/commit/8c637c3b1272be18d8f56c5ac358b28986607e2b)) that I think fixes this and buttons should show up now.

I get kind of an interesting result when I build the latest commit: 

[image]

Here's the output: 
```

WARNING: CPU random generator seem to be failing, disable hardware random number generation
WARNING: RDRND generated: 0xffffffff 0xffffffff 0xffffffff 0xffffffff
Creating application object
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
Creating radeon_profile
Loading configuration
Creating ui elements
Analyzing screen  0
  Analyzing output  0
  Analyzing output  1
  Analyzing output  2
  Analyzing output  3
    Analyzing active mode
      Property is EDID, parsing it
Using QCharRef with an index pointing outside the valid range of a QString. The corresponding behavior is deprecated, and will be changed in a future version of Qt.
Using QCharRef with an index pointing outside the valid range of a QString. The corresponding behavior is deprecated, and will be changed in a future version of Qt.
Using QCharRef with an index pointing outside the valid range of a QString. The corresponding behavior is deprecated, and will be changed in a future version of Qt.
Searching PnP ID:  "AOC"
Initializing device
Card detected:
 module:  "amdgpu" 
 sysName(path):  "card0" 
 name:  "Advanced Micro Devices, Inc. [AMD/ATI] Navi 10 [Radeon RX 5600 OEM/5600 XT / 5700/5700 XT] (rev ca)"
hwmon path:  "/sys/class/drm/card0/device/hwmon/hwmon0/"
Opened /dev/dri/renderD128 , fd number: 15
QFSFileEngine::open: No file name specified
Detected power method based on sysfs power_method file: 3
Power method: Power Profile Modes. Creating profiles list
ioctlHandler: everything ok
ioctlHandler: everything ok
GPU max core clk:  1780 
 max mem clk:  875 
 vram size:  5991.84
Handling found device features
Creating power profiles control buttons
Fan control available

```

marazmista, 2020-07-17:
Thanks for screenshot. I pushed a coomit (https://github.com/marazmista/radeon-profile/commit/673f1cd377936527b2bda8f5db2eece18c506209). I hope it will fix this.

largewedge, 2020-07-18:
> Thanks for screenshot. I pushed a coomit ([673f1cd](https://github.com/marazmista/radeon-profile/commit/673f1cd377936527b2bda8f5db2eece18c506209)). I hope it will fix this.

I've just built the latest commit. The buttons are showing up now.

marazmista, 2020-07-18:
That's great! Thank you very much for testing.

closed 2020-07-28 by largewedge