marazmista/radeon-profile #219: I can't change the "Set power profile to:" and "Set power level:" at event definition's window.

rizzini, 2020-07-15:
When I click on those nothing appear. [Print](https://i.imgur.com/pw8Pyvy.jpg).

My intention is to always keep my GPU on low profile and back it up to normal only when a specified binary is executed, but since I can't select any profile from the "Event definition", I can't find a way to do that. Is it even possible? It seems to be some very basic feature.

marazmista, 2020-07-16:
Yes, that's the idea. You change event trigger to binary and choose from lists profiles that will be applied when you start binary. Are those lists empty (apart from 'No change' item)?
Please take a look at my response here: https://github.com/marazmista/radeon-profile/issues/218#issuecomment-659503026
Do you see buttons like those on the screenshot?

rizzini, 2020-07-16:
> Yes, that's the idea.
I figured, but since nothing appears on those field, I started to question myself about if it's possible or if I am missing something.

> Do you see buttons like those on the screenshot?

No. I don't. Weird.

I have a RX580 8GB.
OS: Arch Linux

Thanks for the reply.

marazmista, 2020-07-18:
#218 has been resolved and lists in Event definition should also work now. Please try build latest commit.

closed 2020-07-18 by rizzini

megatog615, 2022-03-04:
This does not appear to work anymore. I can set Power Profile and Fan but I cannot change Power Level. I am also using an RX580.
[image]