marazmista/radeon-profile #221: (Ubuntu 20.04) Unable to install Radeon Profile

Iron-Squid, 2020-07-18:
Input: 
```
sudo apt install radeon-profile
```

Output:
```
> Reading package lists... Done
> Building dependency tree       
> Reading state information... Done
> Package radeon-profile is not available, but is referred to by another package.
> This may mean that the package is missing, has been obsoleted, or
> is only available from another source
> 
> E: Package 'radeon-profile' has no installation candidate
```

Iron-Squid, 2020-07-18:
I've also attempted `sudo apt install radeon-profile-daemon`.

Output:
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 radeon-profile-daemon : Depends: radeon-profile but it is not installable
E: Unable to correct problems, you have held broken packages.
```

muzena, 2020-07-18:
@Iron-Squid 
From where you install radeon-profile?

Iron-Squid, 2020-07-18:
The terminal..?

muzena, 2020-07-18:
Which repository?

Iron-Squid, 2020-07-18:
Oh, sorry.
I've tried `ppa:trebelnik-stefina/radeon-profile`

muzena, 2020-07-18:
https://launchpad.net/~radeon-profile

Iron-Squid, 2020-07-18:
Success, thank you!

Iron-Squid, 2020-07-18:
Anyone in the future reading this, the repository I used is `ppa:radeon-profile/stable`

muzena, 2020-09-04:
@Iron-Squid 
Can you close this issue since is resolved.

closed 2020-09-04 by marazmista