marazmista/radeon-profile #224: Minor issues and improvements that should be applied to the Graph tab

some-username-here1, 2020-07-19:
As mentioned in the issue's name, plus it would be nice to be able to re-arrange them into any order too. The latter that's mentioned happens when you rename a existing plot and instead of renaming selected plot, it creates a new one.

A suggestion that I would like to add is that the "Apply" button sits to the left of the "Configure" button, and when the graph is being shown, "Apply" would be grayed out; vice-versa for the "Configure" button when you're already at the configuration, and removing the antialiasing that's being applied to the lines on graphs.
Here are some modified screenshots I made that has some of what I'm talking about:
[image]
[image]

marazmista, 2020-07-20:
Hi
I didn't like that plot configuration page either so I'm currently working on moving all of it to separate dialog window but right now it's crashing like crazy during plots modifications/recreation so I need to figure that out.
[image]

some-username-here1, 2020-07-22:
Oh, alright! Nice to know, but what about the buttons for being able to move the listed plots up or down?
Once you figure out what's causing the constant crashes

GaVenga, 2020-07-26:
So crashes are not limited to certain hardware?!

marazmista, 2020-07-27:
> Oh, alright! Nice to know, but what about the buttons for being able to move the listed plots up or down?
Once you figure out what's causing the constant crashes

Those up and down buttons are a bit tricky to implement. Maybe as a workaround for now, it could be possible to force certain order by adding number at the beginning of name.

> So crashes are not limited to certain hardware?! 

What do you mean? If crashes were related to plots(probably yes) it should crash no matter the hardware. But I think it should work fine with latest commits.