marazmista/radeon-profile #226: RX 5500 XT not recognized by radeon-profile

Mango2000, 2020-08-13:
I'm on Manjaro KDE 
The GPU drop down menu is empty. Only the 'conector tab gives some info, but nothing about the gpu. The other tabs are empty

Mango2000, 2020-08-31:
/sys/class/drm/
/sys/class/drm/card0
/sys/class/drm/card0-DP-1
/sys/class/drm/card0-DVI-D-1
/sys/class/drm/card0-HDMI-A-1
/sys/class/drm/renderD128
/sys/class/drm/ttm
/sys/class/drm/version

those one have a driver=amdgpu entry:
/sys/class/drm/card0/device/uevent
/sys/class/drm/renderD128/device/uevent
The other don't have this line in their device/uevent file

Riotvan81, 2020-09-13:
Having the same problem with 20200824. Using a Radeon VII, main os is Gentoo. The daemon is running fine.
DRIVER=amdgpu
PCI_CLASS=30000
PCI_ID=1002:66AF
PCI_SUBSYS_ID=1002:081E
PCI_SLOT_NAME=0000:0d:00.0
MODALIAS=pci:v00001002d000066AFsv00001002sd0000081Ebc03sc00i00
Can't select my gpu. Any suggestions?

Eqlipse, 2020-12-12:
Having an identical issue here, too, running an RX 5700 XT (Sapphire Nitro+). Results for /sys/class/drm are _very_ similar to Mango2000's:

```
/sys/class/drm/
/sys/class/drm/card0
/sys/class/drm/card0-DP-1
/sys/class/drm/card0-DP-2
/sys/class/drm/card0-HDMI-A-2
/sys/class/drm/renderD128
/sys/class/drm/ttm
/sys/class/drm/version
```

And, identically, "DRIVER=amdgpu" shows up in these two locations:
```
/sys/class/drm/card0/device/uevent
/sys/class/drm/renderD128/device/uevent
```