marazmista/radeon-profile #231: Detecting 0 fan speed but the fan curve is working anyway, but not auto. Temperature number works but not pie.

smasher-101, 2020-09-16:
Using Sapphire pulse RX 5700 XT.
When I am on windows my card works fine but on linux it wants to just not spin the fans at all so I got this. Fans still weren't spinning on auto so I set a fan curve and now they are working. However I don't think the temperature reading is working because the temperature pie chart isn't working but the number seems to be. Also the fan speed everywhere it says it claims to be 0 when I can see the fans spinning.

Also when I installed the radeon drivers on windows the fans decided to not work there too and I had to set a curve, however the temperature and fan speed readings seem to be working over there.

smasher-101, 2020-09-16:
Also gpu usage says 0% too.

smasher-101, 2020-09-16:
Now I am getting weird wayland related errors when trying to launch it and when it does launch it doesn't work.

marazmista, 2020-10-04:
Regarding the wrong fan speed it seems like similar issue to this: https://github.com/marazmista/radeon-profile/issues/209 and it was traced back to driver in kernel and is still open: https://gitlab.freedesktop.org/drm/amd/-/issues/1164