marazmista/radeon-profile #233: Overclock / overvolt not working

Razer0123, 2020-09-29:
Used this for months, but if i try to use latest version, my custom profile does not seem to apply anymore

I am running Archlinux with a RX 580 gpu

Using an old version works flawlessly

What could be causing the issue?

marazmista, 2020-10-04:
Do you use this as root or normal user? Is it similar to this issue https://github.com/marazmista/radeon-profile/issues/163?
Can you compile application not for master but for this commit cb3f047e56901b1c5a7907617be33e17d0be538c? You can do this following instructions in build section in readme, but after clone type:
```
cd radeon-profile
git reset --hard cb3f047e56901b1c5a7907617be33e17d0be538c
```

Razer0123, 2020-10-05:
Using as normal user with daemon, yes i suppose it could be the same issue

Just tried, WORKS NOW! :D so happy!

can you push it to master so i can build from AUR?

Thanks for your job, is very appreciated

marazmista, 2020-10-17:
It is in master but it seems like later commits broke something.

Razer0123, 2020-10-23:
So what should we do?

Sonickyle27, 2020-12-09:
I am also being affected by this issue using an RX 580 GPU in my GL702ZC laptop. Using commit [cb3f047](https://github.com/marazmista/radeon-profile/commit/cb3f047e56901b1c5a7907617be33e17d0be538c) allows overclocking profiles to be applied, but version 20200824 compiled from the AUR using `radeon-profile-git` does not work. radeon-profile believes the profile is applied, but `cat /sys/class/drm/card0/device/pp_od_clk_voltage` says otherwise.

Here's some relevant info from neofetch:
```
OS: Manjaro Linux x86_64 
Host: GL702ZC 1.0 
Kernel: 5.9.13-xanmod2-MANJARO 
Shell: bash 5.0.18 
CPU: AMD Ryzen 7 1700 (16) @ 3.000GHz 
GPU: AMD ATI Radeon RX 470/480/570/570X/580/580X/590 
```

If there is any more info you need from me or if you want me to try anything please let me know.